espruino/BangleApps #3493: Daisy clock: Battery time remaining varies wildly, not helpful - please revert or give option to change.

Moini, 2024-07-06:
### Affected hardware version

Bangle 2

### Your firmware version

2v23

### The bug

The latest version of the Daisy clock app uses 'remaining time until 0%' in Days and Hours for displaying the battery capacity. 

This is estimated depending upon the current usage, which means that the values vary wildly and are 100% useless to know when I need to charge the bangle.

Within the last 10 minutes, I had it show anything between 6hours to 120 days. 

Please give us back the percentage indicator - at least as an option. I know what to do with those numbers.
(additionally, the long text really looks ugly, I just need a number there with a percent sign, maybe a tiny icon)

### Installed apps

Sorry, the list has disappeared from the website (why????) and been replaced by a button that holds the app info. I can paste the link here, but I'm not going to rewrite this bug report that I already made (or login with Chromium to GitHub while I'm actually only using it with the Bangle only, because Firefox can't do WebBluetooth).

https://github.com/login?return_to=https%3A%2F%2Fgithub.com%2Fespruino%2FBangleApps%2Fissues%2Fnew%3Ftemplate%3Dbangle-bug-report-custom-form.yaml%26fwversion%3D2v23%26apps%3Dbarometer%2520%280.04%29%250Acalculator%2520%280.07%29%250Acounter%2520%280.04%29%250Afileman%2520%280.03%29%250Agallery%2520%280.03%29%250Ainfo%2520%280.03%29%250Aptlaunch%2520%280.15%29%250Awclock%2520%280.04%29%250Awidlock%2520%280.08%29%250Awidtbat%2520%280.02%29%250Aabout%2520%280.15%29%250Aacmaze%2520%280.05%29%250Aactivityreminder%2520%280.12%29%250Aagenda%2520%280.15%29%250Aandroid%2520%280.36%29%250Acalendar%2520%280.18%29%250Acompass%2520%280.09%29%250Adtlaunch%2520%280.25%29%250Agbmusic%2520%280.12%29%250Agpsinfo%2520%280.13%29%250Ahealth%2520%280.30%29%250Ahrm%2520%280.13%29%250Ainfoclk%2520%280.10%29%250Amessagegui%2520%280.79%29%250Amessageicons%2520%280.07%29%250Amessages%2520%280.62%29%250Amylocation%2520%280.11%29%250Anotify%2520%280.14%29%250Aopenstmap%2520%280.30%29%250Apastel%2520%280.20%29%250Apowermanager%2520%280.11%29%250Aqcenter%2520%280.05%29%250Arandom%2520%280.03%29%250Arecorder%2520%280.43%29%250Arun%2520%280.19%29%250Asched%2520%280.26%29%250Asmpltmr%2520%280.09%29%250Aweather%2520%280.26%29%250Awidalarm%2520%280.02%29%250Awidbt%2520%280.09%29%250Awidmessages%2520%280.06%29%250Awidmp%2520%280.09%29%250Alocale%2520%280.19%29%250Aac_ac%2520%281.00%29%250AUke%2520%280.04%29%250Aswiperclocklaunch%2520%280.07%29%250Aalarm%2520%280.48%29%250Asetting%2520%280.73%29%250Aboot%2520%280.63%29%250Alcars%2520%280.31%29%250Adaisy%2520%280.12%29

Moini, 2024-07-06:
(the list was useful for more purposes than just for making a bug report...)

thyttan, 2024-07-06:
> Sorry, the list has disappeared from the website (why????) and been replaced by a button that holds the app info.

That's weird. The "Apps Installed" list is still there for me on both the official and development app loader.

The "New issue on GitHub" button is there in addition to the list, not as a replacement. I'f you'd want to use it in the future I just verified it's possible to copy the address from chromium and paste it into firefox and you will auto login and load the issue prefilled.

thyttan, 2024-07-06:
> Please give us back the percentage indicator - at least as an option.

The option is there in the settings: Settings->Apps->Daisy->'Expected Battery Life In Hours' (You'll need to toggle it on and off to get battery % - that's a bug). 

@Poolitzer Would you have the time to harmonize the app code and settings code with regards to hours vs percentage left?

Poolitzer, 2024-07-06:
Damnit how did I break that. Will have a look tomorrow sorry

Moini, 2024-07-06:
Ah, I understood it allowed me to switch between hours and days display. Which wasn't what I wanted. Perhaps it would be possible to rename the option so it's clear that it's 'time remaining' vs. 'percentage charged'?

Moini, 2024-07-06:
Confirmed to work now after toggling back and forth.

Moini, 2024-07-06:
> That's weird. The "Apps Installed" list is still there for me on both the official and development app loader.
>
> The "New issue on GitHub" button is there in addition to the list, not as a [...]

Thanks, and sorry. I don't know why I couldn't find it. I did scroll down completely, and even clicked the button to show advanced options in my search for it. Right now, it's there. Very strange.

thyttan, 2024-07-07:
> Thanks, and sorry. I don't know why I couldn't find it. I did scroll down completely, and even clicked the button to show advanced options

Maybe the app loader had failed to fetch the list from the Bangle? That happens to me sometimes.

Moini, 2024-07-07:
@thyttan Thanks, that restores my faith in my sanity a bit :)

Moini, 2024-07-07:
Btw. do you take feature requests for Daisy? I've been looking into the code, and I think I even found the place where I'd need to 'hook the change I'd like in', but ... I don't think I could do it myself in a reasonable amount of time. 
I would like a setting to choose where to start the cycle, so I do not have to tap 6 times to get the date displayed again every time I have opened the menu.

thyttan, 2024-07-07:
You're always welcome to open feature requests on the issue tracker 🙂 

There's no guarantee on if/when a request gets implemented. Depends on if someone else finds it worthwhile.

If you're eager to have it implemented I'd recommend giving it a shot  yourself. You could open a draft pull request to ask for pointers and feedback. 

Some tutorials: 
- https://www.espruino.com/Bangle.js+Development
- https://www.espruino.com/Bangle.js+App+Loader

Poolitzer, 2024-07-07:
@Moini feel free to ping me on 05th if you want me to implement your feature request, till then quite busy

closed 2024-07-07 by thyttan

Moini, 2024-07-08:
Thanks all for the help! 

As for the feature request, I don't think I'll find the time to learn enough Bangle scripting to make this work, but I've opened an issue about it.

PR #3495 (2024-07-07): Daisy Fix: Put default to False for battery display
This fixes #3493
commit e47c1c34 (2024-07-07): Daisy Fix: Put default to False for battery display

This fixes #3493