espruino/BangleApps #3514: [App loader] Create "development" category?

pavelmachek, 2024-07-16:
Some tools are useful for people doing development on B.js2, but not much for anyone else. Having "devel" or "development" category would make "tools" less cluttered and such apps would be easier to find.

bobrippling, 2024-08-02:
Sounds good - @gfwilliams wdyt?

gfwilliams, 2024-08-05:
Sounds good to me too. Lets call it `devel` and then label `Development` in the app loader (I think it needs mentioning in the readme too).

When we add it I'd like to ensure we go through and tag any existing apps that fit so there's actually something in the menu - maybe you could suggest the apps you think should go in @pavelmachek?

pavelmachek, 2024-08-08:
"iconbits" and "forge" should definitely go there. Plus I guess various power debugging apps/widgets could go there, as they'll mostly be used for application development. Probably "Accelerometer Graph (v0.01)", "Gadgetbridge Debug (v0.01)", "Messages Debug (v0.01)" and "Sensor tools (v0.06)" fit into that category as well.

closed 2026-11-13 by gfwilliams