espruino/BangleApps #3523: [messages] Music message never gets deleted

kurnevsky, 2024-07-25:
### Affected hardware version

Bangle 2

### Your firmware version

2v23

### The bug

After I launch a music player on my phone and switch several songs or just pause/play several times some music message get persisted in storage and is never deleted. I always see it in the messages list, and more importantly whenever I start playing a new song/switch to a different song this new song is only shown for a very short amount of time in the messages ui after which I again see the old saved message with wrong old title. It looks like I have 2 independent music messages and one is displayed over another.

Koell, 2024-10-07:
I have that problem with multiple types of messages
[issues/3603](https://github.com/espruino/BangleApps/issues/3603)

bobrippling, 2024-10-07:
Tagging @gfwilliams with this being a more central app

gfwilliams, 2024-10-08:
Please can you paste up some Gadgetbridge logs of when this happens? https://gadgetbridge.org/internals/topics/logs/

This'll help to track down exactly where the issue lies and to reproduce it - whether it's something in Gadgetbridge on the phone, or on the Bangle.

Note: the logs may have personal info if you received any messages during the time you were making the log, so it's work opening the file and deleting/redacting any personal info you see.

gfwilliams, 2024-10-08:
I've made some changes to the messages app now - there was a bug where the 'Music' message was never marked as unread so it always popped up, and that is now fixed.

I also added a button at the top right that allows you to get to the messages menu (like we normally have on messages) where you can delete the music message completely if you just don't want it there.

So I think that's most things fixed now - unless you can get me a log message showing the swap from the old/new music message

kurnevsky, 2024-10-08:
It's still reproducing for me. And since it works for most cases I have a suspicion it might be caused by `kde connect` notifications and playing videos from youtube with plasma integration.

> Which Gadgetbridge are you using? Play Store, or the F-Droid one?

F-Droid one.

> In the Android app on the watch, is Keep Msgs set?

It's not set.

> I also added a button at the top right that allows you to get to the messages menu (like we normally have on messages) where you can delete the music message completely if you just don't want it there.

Is it in released version already? I don't see such button :)

> Please can you paste up some Gadgetbridge logs of when this happens?

Here is `grep UART` from logs:

```
23:37:30.785 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: setTime(1728423450);E.setTimeZone(2.0);(s=>s&&(s.timezone=2.0,require('Storage').write('setting.json',s)))(require('Storage').readJSON('setting.json',1))
23:37:30.785 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"is_gps_active"})
23:37:31.057 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX LINE: 
23:37:31.104 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX LINE: {"t":"gps_power","status":false}
23:37:31.104 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX JSON parsed successfully
23:37:32.669 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX LINE: 
23:37:32.670 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX LINE: {"t":"status","bat":35,"chg":0}
23:37:32.672 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX JSON parsed successfully
23:37:32.673 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX LINE: 
23:37:32.673 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX LINE: {"t":"ver","fw":"2v24","hw":2}
23:37:32.674 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX JSON parsed successfully
23:37:32.689 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX LINE: 
23:37:32.701 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX LINE: {"t":"force_calendar_sync","ids":[]}
23:37:32.701 [binder:6556_1] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART RX JSON parsed successfully
23:37:39.692 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"notify","id":1728422415,"src":"KDE Connect","title":"KDE Connect","subject":"","body":"some video","sender":""})
23:37:39.753 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicinfo","artist":"some artist","track":"some video","dur":16353,"c":-1,"n":-1})
23:37:39.755 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicstate","state":"pause","position":13158,"shuffle":-1,"repeat":-1})
23:37:42.471 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"notify-","id":1728422415})
23:37:46.785 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicinfo","artist":"some artist","track":"some video","dur":16353,"c":-1,"n":-1})
23:37:46.787 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicstate","state":"play","position":13158,"shuffle":-1,"repeat":-1})
23:37:47.051 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"notify","id":1728423316,"src":"KDE Connect","title":"KDE Connect","subject":"","body":"some video","sender":""})
23:37:47.135 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicinfo","artist":"some artist","track":"some video","dur":16353,"c":-1,"n":-1})
23:37:47.137 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicstate","state":"pause","position":13160,"shuffle":-1,"repeat":-1})
23:37:49.828 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"notify-","id":1728423316})
23:37:53.100 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicinfo","artist":"another artist","track":"another video","dur":1819,"c":-1,"n":-1})
23:37:53.101 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicstate","state":"pause","position":57,"shuffle":-1,"repeat":-1})
23:38:00.095 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicinfo","artist":"some artist","album":"some track","track":"some track","dur":547,"c":-1,"n":-1})
23:38:00.098 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicstate","state":"play","position":333,"shuffle":-1,"repeat":-1})
23:38:03.285 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicinfo","artist":"some artist","album":"another album","track":"another track","dur":313,"c":-1,"n":-1})
23:38:03.287 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicstate","state":"play","position":0,"shuffle":-1,"repeat":-1})
23:38:17.041 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicinfo","artist":"some artist","track":"some video","dur":16796,"c":-1,"n":-1})
23:38:17.043 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicstate","state":"play","position":13196,"shuffle":-1,"repeat":-1})
23:38:17.173 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicinfo","artist":"some artist","track":"some video","dur":16796,"c":-1,"n":-1})
23:38:17.175 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicstate","state":"play","position":13196,"shuffle":-1,"repeat":-1})
      24
[... cut ...]
23:38:47.652 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicstate","state":"play","position":0,"shuffle":-1,"repeat":-1})
23:38:51.471 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicinfo","artist":"ignored artist","album":"ignored album","track":"ignored track 2","dur":284,"c":-1,"n":-1})
23:38:51.472 [main] INFO  n.f.g.s.d.b.BangleJSDeviceSupport - UART TX: GB({"t":"musicstate","state":"play","position":0,"shuffle":-1,"repeat":-1})
```

It got stuck on `another track` (replaced name) from the logs, and `ignored track`s aren't shown anymore. Should I include something else from the logs?

thyttan, 2024-10-08:
> Is it in released version already? I don't see such button :)

It's on the development app loader. You can access it from the "More..." tab of the official app loader.

gfwilliams, 2024-10-09:
Thanks for the log!

Ok, so it definitely looks like `KDE Connect` is popping up notifications with the wrong video in, and then Gadgetbridge is parsing those and only showing the 'correct' video when they go away.

What happens if you go into the Gadgetbridge Blacklist Notifcation window and just block notifications from KDE Connect? Does it work ok then? However looking at Gadgetbridge source, it listens to `com.android.music.playstatechanged`/etc and if the music info is coming through those there's really not much we can do. Android thinks the currently playing file is changing, so that's what we're forwarding,

If KDE Connect is erroneously overwriting the currently playing song on your phone, that's an issue you should file with them,