espruino/BangleApps #3534: [locale] Improve support for meridians

atjn, 2024-08-15:
Right now it is impossible to properly implement support for meridians on clocks. You can use `require("locale").meridian(new Date())` to get the current meridian and display that somewhere in your clock. This will work in some cases but not in the following cases:

1. The user's locale includes meridians but the user chose `24 hour clock` in settings. In this case, you will still be showing the meridians even though you are displaying a 24-hour clock. To remove the meridians, you must manually read the settings file and check if the setting is set to 12-hour or 24-hour, then you must stop displaying the meridian based on that information. That is not a very nice solution.
2. The user's locale does not include meridians but the user chose `12 hour clock` in settings. All locales that traditionally use 24-hour clocks do not have meridian information. If the user chooses a 12-hour clock here, you have no way of displaying a meridian.

Another issue is that there is no documentation on how to use the meridians. It seems like most clocks have simply opted to never show meridians, probably because they were not aware of it, or because they couldn't figure out how to use it properly.

I think we need the following changes to properly support meridians:

- [ ] `locale` should return an empty string for meridian when the user has chosen a 24-hour clock. (fixes problem 1)
- [ ] Include meridian names in all locales. (fixes problem 2)
- [ ] Document how to use meridians correctly in [Bangle.js Internationalisation](https://www.espruino.com/Bangle.js+Locale).
- [ ] Add an `is12Hour` function to `locale` so apps can easily know which clock format the user has chosen.

The last one is not as important as the others, but I could imagine that some apps would want to change the layout depending on the clock format.

What do you think?

bobrippling, 2024-08-19:
Tagging @gfwilliams with this being a central change :)

gfwilliams, 2024-08-27:
This seems like a sensible change - just merged those two changes. If you could contribute those improved docs you mention that'd be great.

One thing I was wondering is what if a clock/app **wants** to get meridian regardless of if 24 hour is specified or not? We'd have the data in the `locale` module but would then refuse to give it to them.

Maybe instead we should actually have `meridian: (d,force) => (force||is12Hours()) ? d.getHours() < 12 ? ${js(locale.ampm[0])}:${js(locale.ampm[1])} : "",
  is12Hours,` so at least it's possible to get meridian.