espruino/BangleApps #3546: [gipy] Bigger area fails to covert

pavelmachek, 2024-08-27:
Smaller area works ok (resulting in 130K file). But when trying to prepare for 1-hour car trip (and slightly bigger area), I press "convert", then get warning about page not responsive, and then nothing.

Would it be possible to upload _just_ the gpx track to watch (without surrounding map), or maybe have low fidelity map with just big towns and major roads for such use?

Musaveershaik, 2024-10-28:
Kindly assign me this task under Hacktoberfest2024, Thanku.

thyttan, 2024-10-28:
> Kindly assign me this task under Hacktoberfest2024, Thanku.

@gfwilliams @bobrippling @halemmerich do any of you know how hacktoberfest works?

gfwilliams, 2024-10-28:
I think ideally people just submit PRs with fixes in, but it's easy enough for us to add the 'hacktoberfest' label. I've assigned you too @Musaveershaik but generally if we're working on something and it might take a while we just post a comment up rather than assigning