espruino/BangleApps #3581: Out of memory when using OpenTrack

dror3go, 2024-09-21:
### Affected hardware version

Bangle 2

### Your firmware version

2v24

### The bug

Using OpenTracks app while the phone is connected to the Banglejs2 watch - at some point the phone turns blank with a blurry message at the bottom starting with `FIFO_EOWL` and ending with `MEMORY`. The rest is overlapping letters which makes it impossible to read.
From that point - the phone becomes useless: long press on the button reloads it, but it doesn't update the clock, it doesn't show notifications, and the screen is not interactive.

bobrippling, 2024-09-23:
Which is the OpenTracks app? Do you mean [Open Street Map](https://banglejs.com/apps/?id=openstmap)?

dror3go, 2024-09-23:
No, I mean the [Android OpenTracks app](https://github.com/OpenTracksApp/OpenTracks), which probably generates many notifications on the Banglejs watch during activity.
I ended up doing a factory reset on the watch since it became unusable.

bobrippling, 2024-09-23:
Gotya - sounds like a problem with either [Android](https://banglejs.com/apps/?id=android) or the [Messages App](https://banglejs.com/apps/?id=messages) (if you're using the default messages app?)

@gfwilliams what do you think? Notification throttling?

gfwilliams, 2024-09-24:
I think this may have come up before - as I understand it it's kind of an issue with OpenTrack - it creates literally hundreds of notifications (I think when it's downloading something or displaying some kind of progress?). I *think* other apps use 'local' notifications which get ignored by Gadgetbridge but OpenTrack makes them global.

You can go into Gadgetbridge, notification settings and block all notifications from OpenTrack and I think that'll fix it though. It shouldn't have been anything wrong with the Bangle itself - once Gadgetbridge disconnected and stopped throwing notifications at it I think it would have deleted the existing notification file and it all would have been fine.

We could try and push something to Gadgetbridge that blocks OpenTrack by default - I guess throttling is possible but it sounds problematic - someone's going to complain that something didn't get through

gfwilliams, 2025-10-23:
Interesting, thanks! So this would be a Gadgetbridge issue then - I wonder whether it needs to throttle notifications somehow, or maybe there's a specific notification type it should be ignoring.

Are you able to get a log from Gadgetbridge for when this is happening?

It's possible that because Espruino is able to respond to notifications that change (and I think a lot of other Gadgetbridge device handlers don't) we're getting hammered by it. We could potentially just ignore notification changes on Signal and OpenTracks