espruino/BangleApps #3589: [Messages vs Music Control] vibration pattern doesn't stop

aemkai, 2024-09-26:
### Affected hardware version

Bangle 2

### Your firmware version

2v24

### The bug

With active (loaded) music control app, a notification alarm doesn't stop after the configured timeout, instead the pattern is repeated until the music app is closed by pressing hw-button two seconds.
Short Pressing the button only changes play/pause state

- It doesn't matter if music is playing or not when the message is received.
- It doesn't matter if the music control app is started automatically or by hand.
- It doesn't matter if "Auto-Open Music" is dis- or enabled.
- Configuration of Messages UI seems to make no difference.

MessagesUI  is configured as follows:
- Vibrate ::
- Vibrate for calls ::
- Repeat 5s
- Repeat for calls 5s
- Vibrate timer Off
- Unread timer 10s
- Min Font Small (default)
- Auto-Open Music [ ]/[v] (both tested)
- Unlock Watch [v]
- Flash Icon [ ]
- Quiet Mode disables auto.open [ ]
- Disable auto-open [ ]
- Widget Messages 3 (default)
- Icon color mode color (default)
- Car driver pos lef (default)

### Installed apps

Message UI (v0.80)
Message Widget (v0.06)
Messages (v0.62)
Message Icons (v0.08)
Gadgetbridge Music Controls (v0.13)

---
About
Lock Widget
Battery Level Widget
Welcome
Notifications (default)
Health Tracking
Scheduler
Alarms Widget
Alarm & Timers
Android Integration
Setting
Bootloader
Sliding Clock
Clock Info Module
Binary LED Clock
Desktop Launcher
2047pp
Languages

some custom clock apps in development

bobrippling, 2024-09-30:
At a guess, the showPrompt might be being interfered with by the `setTimeout` juggling that the GB music code is doing - if you have [the IDE](https://www.espruino.com/ide/) connected while reproducing the bug, do you see any output in the console?

aemkai, 2024-10-01:
I'm actually short of time, I will try to test this on Thursday.

aemkai, 2024-10-03:
Okay, seems that I'm to stupid to connect my Bangle in web-IDE, while it is connected with Gadgetbridge, even if I use the same device. Is this even possible? 
And is there a way to control GB music and Messages from IDE/console?

bobrippling, 2024-10-03:
Ah no, you're right - you can't connect more than one thing to the Bangle at once, this will be more gnarly to debug. @gfwilliams before I dive into somehow catching errors and saving to flash, do you have any techniques you've used for this already?

thyttan, 2024-10-03:
I think this should work: https://www.espruino.com/Gadgetbridge#remote-debugging

gfwilliams, 2024-10-04:
Yes, as @thyttan says - I added it for exactly this eventuality :)

aemkai, 2024-12-28:
Long last is finally good, or even not. Sorry for the long break, but I was short of time.
Actually I tested the issue (and setting up the connection was a bit shaky), but finally the following output was displayed on the terminal (comments in square brackets):

```
**[music is playing, alarm is set up, then connecting via Remote Connection Bridge]**

Unable to retrieve board information. Connection Error?
Connected to Android (No response from board)
>< << {"VERSION":"2v24","GIT_COMMIT":"9b3cc7f80","BOARD":"BANGLEJS2","RAM":262144,"FLASH":1048576,"SPIFLASH":8388608,"HWVERSION":2,"STORAGE":8388608,"SERIAL":"...","CONSOLE":"Bluetooth","MODULES":"Flash,Storage,heatshrink,crypto,tensorflow,locale","EXPTR":671884,"APP_RAM_BASE":536884832} >> >
notify {
  "t": "notify",
  "id": ...,
  "src": "Kalender",
  "title": "test",
  "subject": "",
  "body": "12:07",
  "sender": ""
 }
>BLE Connected, queueing BLE restart for later

**[next try, stop connection, music is playing, alarm is set up, then re-connecting via Remote Connection Bridge]**

Found BANGLEJS2, 2v24
>
New Firmware available (2.24 installed, 2v25 available)
Connected to Android
>BLE Connected, queueing BLE restart for later
BLE Connected, queueing BLE restart for later
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v24 (c) 2024 G.Williams
notify {
  "t": "notify",
  "id": ...84,
  "src": "Kalender",
  "title": "test",
  "subject": "",
  "body": "12:16",
  "sender": ""
 }
**[vibration is ongoing, pressing button long]**

>Button held down - interrupting JS execution...
Execution Interrupted

**[starting music again on mobile, setup new alarm]**

>BLE Connected, queueing BLE restart for later
BLE Connected, queueing BLE restart for later
BLE Connected, queueing BLE restart for later
BLE Connected, queueing BLE restart for later
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v24 (c) 2024 G.Williams
notify {
  "t": "notify",
  "id": ...85,
  "src": "Kalender",
  "title": "test",
  "subject": "",
  "body": "12:18",
  "sender": ""
 }
**[vibration is ongoing till alarm is acknowledged on mobile, then it stops. Set up new alarm]**

notify {
  "t": "notify",
  "id": ...86,
  "src": "Kalender",
  "title": "test",
  "subject": "",
  "body": "12:19",
  "sender": ""
 }
**[vibration is ongoing]**
>
{"t":"act","stp":0,"hrm":0,"mov":141}
{"t":"music","n":"pause"}
{"t":"music","n":"play"}
{"t":"music","n":"pause"}
{"t":"music","n":"play"}
{"t":"music","n":"pause"}
notify {
  "t": "notify",
  "id": ...87,
  "src": "Kalender",
  "title": "test",
  "subject": "",
  "body": "12:21",
  "sender": ""
 }
>
**[vibration is ongoing]**

{"t":"music","n":"play"}
{"t":"music","n":"pause"}

**[stop music app on watch, setup new alarm on mobile]**

BLE Connected, queueing BLE restart for later
BLE Connected, queueing BLE restart for later
notify {
  "t": "notify",
  "id": ...88,
  "src": "Kalender",
  "title": "test",
  "subject": "",
  "body": "12:23",
  "sender": ""
 }
>BLE Connected, queueing BLE restart for later
BLE Connected, queueing BLE restart for later
**[vibration is stopping (without music-app on watch)]**
 
```