espruino/BangleApps #3593: [widbt_notify x lcdclockplus] clockinfo is not redrawn immediately after go back from widget notification to clock

kamilkrz, 2024-09-28:
### Affected hardware version

Bangle 2

### Your firmware version

2v24

### The bug

Hi,
As I dont  think there is replacement for lost bt notification I would like to mark this as occurring issue. 
As messages ui notification is not breaking returning to watchface (only other popup overlay that I am using) I would investigate this a but further but any help would be appreciated. I see that this widget is a bit dated now and probably now there are more clear ways to go back from popup to watchface.

[image]
[image]
As shown clockinfo is redrawn only when its call its emit redraw (timers, events etc.) and not immediately after returning from notification popup.

Thanks in advance

bobrippling, 2024-09-30:
Sounds like a bug with `widbt_notify` or the app - looking at the code, it's both.

`widbt_notify` [expects a global `draw` function to call](https://github.com/espruino/BangleApps/blob/a32d268b7ddb087fff32b8131f4feda86511dd33/apps/widbt_notify/widget.js#L42-L44), which should redraw the widgets (this isn't a hard contract, just implicit). `lcdclockplus` [has a global `draw` function](https://github.com/espruino/BangleApps/blob/a32d268b7ddb087fff32b8131f4feda86511dd33/apps/lcdclockplus/app.js#L14-L14), but doesn't redraw clkinfo.

I would say `widbt_notify` should avoid this implicit coupling and redraw clkinfo itself - via `require("clock_info").clockInfos` - @KungPhoo / @pebl-hank this is perhaps one for you, or we can see who picks this up

gfwilliams, 2024-10-28:
Just fixed this by using the 'notify' library so if you update now it should be ok.

I also spotted 2 other widgets that completely clear the screen and fixed those.

The slightly odd thing is this should only attempt a redraw if there is `draw` function, which there shouldn't be for lcdclockplus - so there must be some other widget/bootcode/etc that is broken that's defining a global `draw` function and fooling it.

closed 2024-10-28 by gfwilliams