espruino/BangleApps #3596: App Loader - info file missing/incorrect - when not using lowercase storage names

bmsleight, 2024-09-29:
Solution - app develops always use lowercase in metadata.json storage names. :) 

Storage name is used from metadata.json - but info file generated is always lower case. 
e.g. {"name":"ExactWords.app.js","url":"app.js"} generates exactwords.info
Clock app appears to be successfully loaded, is shown under Settings->System menu->Clock but can not be selected.

See attached JavaScript console output from forked app loader. 
[console.txt](https://github.com/user-attachments/files/17181311/console.txt)

gfwilliams, 2024-09-30:
Do you have a link to a repo where this is happening so I can take a look? From https://github.com/espruino/BangleApps/compare/master...bmsleight:BangleApps:master it feels like maybe you have fixed it now?

I just checked the code and it looks like `id` is passed through ok, and I imagine from `showMessage "Erasing\nexactwords..."` it thinks the app ID really is `exactwords`.

So I wonder if maybe the app ID is set to `exactwords` but you've named the file `ExactWords.app.js`? I'm surprised the test code built to run on each GitHub commit didn't flag that up?

bmsleight, 2024-09-30:
Hi Gordon,

Thanks

>So I wonder if maybe the app ID is set to exactwords but you've named the file ExactWords.app.js? I'm surprised the test code built to run on each GitHub commit didn't flag that up?

Yep - that where I made the error.   I was not consistent in metadata.

```
 "id": "exactwords", 
....
  "storage": [
    {"name":"ExactWords.app.js","url":"app.js"}, 
```

I have corrected, all the errors were at my end, not in app loader.

closed 2024-09-30 by bmsleight

gfwilliams, 2024-10-01:
Great, glad you got it sorted!