espruino/BangleApps #3603: [Message] Messages are not removed

Koell, 2024-10-07:
### Affected hardware version

Bangle 2

### Your firmware version

2v24

### The bug

Some messages are kept indefinitly, currently I can't narrow down this behaviour

A possible fix would be keep messages on the bangel for a maximum of a day

gfwilliams, 2024-10-08:
Please can you paste up some Gadgetbridge logs of when this happens? https://gadgetbridge.org/internals/topics/logs/

This'll help to track down exactly where the issue lies and to reproduce it - whether it's something in Gadgetbridge on the phone, or on the Bangle.

Note: the logs may have personal info if you received any messages during the time you were making the log, so it's work opening the file and deleting/redacting any personal info you see.

gfwilliams, 2024-10-08:
Also: Which Gadgetbridge are you using? Play Store, or the F-Droid one?

Koell, 2024-10-08:
The F-Droid version, log follows as soon as this occurs again

closed 2024-10-09 by Koell