espruino/BangleApps #3617: Build Layout.js lib into Bangle.js firmware

gfwilliams, 2024-10-21:
Right now, every app that uses Layout has to include its own copy. Parsing it in initially takes 25ms by itself, and then executing it to render on something like a navigation message takes 240ms when running from internal flash would be 220ms.

... so if we built it into the Bangle.js firmware we'd save around 45ms loading something like a navigation message.

The reason we didn't build it in initially was that it was changing rapidly, but the last real change was made a year ago so I think we'd be safe.

Of course if it were to need changes after this, realistically we're looking at waiting until the next Bangle.js firmware release before those changes could go it...

Any thoughts?

gfwilliams, 2024-10-21:
Just to add to this, we also have some `"ram"` keywords to aid performance from external flash. When running on internal flash those could be removed and would likely get it running even faster.

bobrippling, 2024-10-23:
> every app that uses Layout has to include its own copy

Could we get around this by loading it on as a module, like what we do with `clock_info`, So the JS file sits on the device itself, rather than a `Modules.addCached(...)` ?

But I suppose that wouldn't address the performance, and I'd be please with a 25ms speed up, so +1 from me on that.

gfwilliams, 2024-10-24:
Yes, it could sit on the device but you don't really get all the speed benefits then.

Ok, just done this with ef5c7c8a38868d50fdea5fa32d04ba540b1e3c20 - but it needs an updated BangleApps to realise that Layout exists and to not re-upload it

I also improved some other stuff, but now to get Messages to the point where it displays a message we're down from 331ms in 2v24 to 308ms in 2v24.128

Not quite as much as I'd have liked, but still good!

closed 2024-10-24 by gfwilliams