espruino/BangleApps #3623: [Run]  1st screen broken UI

kachna, 2024-10-25:
### Affected hardware version

Bangle 2

### Your firmware version

2v24

### The bug

When it is started the Run/Run+ (both affected) application shows blank screen with dots on left side. It looks that app is somehow working. When I push the button dots react and start changing.
[image]

After swipe right I can see 2nd screen with heart monitor which looks well.
[image]

### Installed apps

sched (0.26), alarm (0.48), setting (0.72), boot (0.62), antonclk (0.11), about (0.15), widlock (0.08), widbat (0.12), widbt (0.09), widid (0.03), welcome (0.14), notify (0.14), health (0.30), widalarm (0.02), gpsinfo (0.13), hrm (0.13), gpstime (0.05), files (0.08), runplus (0.27), recorder (0.44), openstmap (0.30), poweroff (0.04), dtlaunch (0.25), ptlaunch (0.15)

bobrippling, 2024-10-26:
Yeah I'd say this is related to the layout change (see #3617)

gfwilliams, 2024-10-28:
Fixed now, sorry about that. If you re-update the run app from the development app loader it'll be fine

closed 2024-10-28 by gfwilliams