espruino/BangleApps #3634: "Execution Interrupted during event processing" - interrupts clock apps

bobrippling, 2024-11-01:
### Affected hardware version

Bangle 2

### Your firmware version

2v24

### The bug

### Describe the bug

The currently active app appears to be frozen, I suspect its code has run and been interrupted before it's been able to install a `setTimeout()`

I see `Execution Interrupted during event processing.` in the IDE when connecting.

### Expected behavior

The app isn't interrupted and continues the render / perform its tasks

### Steps to reproduce

1. Run (for example) [slopeclock++](https://espruino.github.io/BangleApps/?id=slopeclockpp)
2. Wait some time
3. Observe the clock freezes
4. IDE presents the above message

### Installed apps

Can't establish a connection from the app loader - execution is interrupted before it manages to fetch the app list. With the IDE also connected I see:

```
[Execution Interrupted
>[Execution Interrupted
>[Execution Interrupted
>[Execution Interrupted
>[Execution Interrupted
> 
```

bobrippling, 2024-11-01:
Tagging @gfwilliams too - have you seen anything like this before?

gfwilliams, 2024-11-04:
Maybe try running `E.getErrorFlags` in the IDE? It might be it's failing because the Bangle is out of memory, but it only tends to report the error to the console once - so if you weren't connected when it was reported you might miss it unless you check?

bobrippling, 2024-11-04:
Thanks - I've installed this boot code, stashing here in case anyone else stumbles on this:

```javascript
E.on("errorFlag", function(errorFlags) {
  const f = require("Storage").open("errorflags.log", "a");

  f.write(Date.now() + " " + JSON.stringify(errorFlags) + "\n");

  console.log("errorflags.boot.js: wrote to errorflags.log");
});
```

Will see what I end up with

bobrippling, 2024-11-07:
Looks like OOMs, only occurs when I have GadgetBridge connected (Android), investigating further