espruino/BangleApps #3639: [Message UI] Updating to v0.81 impossible on Bangle.js 1

StuntsPT, 2024-11-04:
### Affected hardware version

Bangle 1

### Your firmware version

2v24

### The bug

Trying to update `Message UI` from v0.79 to v0.81 fails with the following message:

`Message UI update failed, Error: Dependency of 'reply' listed, but nothing satisfies it!`

I suspect that is because the "reply" app is only available for the Bangle.js 2.
Removing "reply" from [here](https://github.com/espruino/BangleApps/blob/7da28950e650210102766ea3ccefa79edd9a58d5/apps/messagegui/metadata.json#L11) solves the issue on my [fork](https://github.com/StuntsPT/BangleApps), I am, however, not sure how to solve this at an official level as to keep "reply" as a dependency for the Bangle.js 2 only.

thyttan, 2024-11-04:
Ping @gfwilliams for visibility.

I wonder if reply lib would work on Bangle 1 and it's just not tested @WhelanB ?

WhelanB, 2024-11-04:
Ah, I don't think I realised this would block on installing Message UI - I targeted Bangle 2 as I wasn't sure of the keyboard support on Bangle 1, but even without a keyboard the app should still function just using canned responses from the interface.html or Gadgetbridge nightly, so I see no reason why it wouldn't work on Bangle 1 - the UI itself is just a default ``showMenu()`` call

thyttan, 2024-11-04:
`kbmorse` is available for the Bangle.js 1 👌

WhelanB, 2024-11-04:
I just gave it a run through on the Bangle 1 emulator via the IDE and everything seems to be functioning ok - happy to open a PR to update the metadata if needs be

WhelanB, 2024-11-04:
I've gone ahead and opened a PR here:
https://github.com/espruino/BangleApps/pull/3640

gfwilliams, 2024-11-04:
Thanks! Just merged.

IIRC in messages we check to see if the reply lib exists and only offer replies if so? Not sure when others' thoughts are but I guess we could remove the dependency? It would mean folks would manually have to install the reply lib if they wanted it, but on Bangle.js where we're very tight for RAM it's possible it might be quite useful - maybe wait and see if anyone has  issues with it

WhelanB, 2024-11-04:
Yes, that's true - it'll only show the reply options if reply is installed, so removing it as a dependency should be no problem if issues start to crop up on Bangle.js - I think it helps to have as a dependency if there are no issues, as it means users will get the reply functionality without having to discover it for themselves!

StuntsPT, 2024-11-06:
I have now updated `Message UI` to the latest v0.82, which successfully pulls the `Reply` app as a dependency.
I am going to test it for the next few days and if any issues crop up I will request a reopen.
The symptom that led me to opening this issue, however, is fixed.

closed 2024-11-06 by StuntsPT