espruino/BangleApps #3648: [ContourClock] custom.html could remember previous font choices when updating?

fanoush, 2024-11-07:
### Affected hardware version

Bangle 2

### Your firmware version

2v24

### The bug

Due to the font selection via custom.html this app cannot be autoupdated from the apploader.
Also every time I update it manually I forget which fonts I have selected previously so I must go over the fonts and guess/pick them again.

Is it possible that the custom.html would read settings from the installed app and preselected my previous fonts and possibly even allowed autoupdate when I do not want any change? I am not sure it can be done, maybe @gfwilliams knows?

gfwilliams, 2024-11-07:
> Is it possible that the custom.html would read settings from the installed app and preselected my previous fonts 

Yes, it's fine. You can do this in custom.html:

```
     function onInit() {
          Util.readStorageJSON('stuff.json', currentSettings => {
```

Just set customConnect:true to force Bangle.js to be connected before upload: https://github.com/espruino/BangleApps/blob/c217a4f858e22d856521a84f7cf18ca728a5ac5e/README.md?plain=1#L302-L305

... but there's nothing to allow auto-update of customised apps yet. That would probably be a bit of a pain to implement

fanoush, 2024-11-08:
Thanks, @xxDUxx  do you want to look into this or may I try? I have very little experience with changing bangle apps so far but I can try.

xxDUxx, 2024-11-09:
Sure, go ahead!

closed 2026-11-12 by bobrippling