espruino/BangleApps #3658: [CirclesClock] Display bug on circles

girardcarillo, 2024-11-12:
### Affected hardware version

Bangle 2

### Your firmware version

2v24.170

### The bug

Sometimes some code lines appear on the main screen instead of one (or several) of the circles (see picture).
When I go to Settings (by pressing the button), and I go back to the main screen (by pressing again the button), the bug disappears.
As you can see on the picture, it looks like the "step" circle (the one in the middle here) is still showing, but above the code lines.
[image]

### Installed apps

On top of Circles Clock:
About, Welcome, Notifications (default), Bootloader, Settings, 
Lock Widget, Health Tracking, Alarm Widgets, Scheduler, Clock Info Module, Clock Backgrounds, Android Integration, Recorder, Desktop Launcher, Bluetooth States, Battery level Widget (with percentage), Temperature Widget

girardcarillo, 2024-11-12:
Follow up: even more code lines appear if I turn off the Bluetooth from my phone (so the watch disconnects from the phone)

[image]

girardcarillo, 2024-11-12:
Ok so I'm changing the name of this issue, as I have the same with Pebble++ Clock (see screenshot)
The bug seems to always appear when I switch on and/or off the Bluetooth of my phone.

[image]

bobrippling, 2024-11-12:
Thanks for the report - if you attach to your watch with [the IDE](https://www.espruino.com/ide/) and trigger the error, could you paste the output here? (selecting it will automatically copy it from the IDE's console)

signaleleven, 2024-11-14:
I struggled with this too, until yesterday.
I first attempted to disable debug logging on display (Settings, utils, debug -> Off, or Log) but nothing helped.

I then randomly tried (while debug was set to "Log") "Rewrite Settings" (Settings, utils, rewrite settings) and since then my screen has been clean, despite trying repeatedly to trigger the output by disconnecting bluetooth (I assume other out of memory errors or so still happen too, I just don't see them and they recover)

Not sure if writing settings has to be done explicitly, or this is a bug.

Hope this helps, and it's not spreading magical thinking.

girardcarillo, 2024-11-14:
Hi @bobrippling thanks for your quick answer.
Note that it was my first time using the WebIDE, I may have done things wrong.

My configuration:
I use my android phone with Chrome via Bluetooth to attach the Bangle to the webIDE (the plan is to do it with my laptop and not my phone bc it's annoying, but in the moment I can't so let's use the phone).
And I am using GadgetBridge on my phone.

The bug shows either
1. [Bangle **not connected** to webIDE] when I connect/disconnect to my Bangle via GagdetBridge
2. [Bangle **not connected** to webIDE] when I turn on/off the phone's Bluetooth (equivalent to 1.)
3. [Bangle **connected** to webIDE] when I turn off the Bluetooth on my phone. In this case, here is the output in the webIDE console:
```
Found BANGLEJS2, 2v24
>
Connected to Web Bluetooth, Bangle.js e326
>
{"t":"status","bat":47,"chg":0}
{"t":"ver","fw":"2v24","hw":2}
{"t":"force_calendar_sync","ids":[]}
Disconnected from Web Bluetooth, Bangle.js e326
```

Two things:
- If Bangle is attached to the webIDE, and I connect and disconnect GB, **the graphical bug doesn't show on the Bangle**, unfortunately. But I still give the output of the IDE's console:
```
Found BANGLEJS2, 2v24
>
Connected to Web Bluetooth, Bangle.js e326
>
{"t":"status","bat":42,"chg":0}
{"t":"ver","fw":"2v24","hw":2}
{"t":"force_calendar_sync","ids":[]}
{"t":"status","bat":47,"chg":0}
{"t":"gps_power","status":false}
{"t":"act","stp":0,"hrm":96,"mov":319}
{"t":"act","act":"activity","ts":1731586800000}
Disconnected from Web Bluetooth, Bangle.js e326
```
- As I said in 3., I can still trigger the bug by connecting to the webIDE, and then turning off the Bluetooth on my phone. But in that case, I guess that, as you disconnect the watch, you don't have any useful output in the IDE's console.

girardcarillo, 2024-11-14:
Hi @signaleleven, thanks for your input.
Unfortunately, I tried what you proposed, and I still have the bug :'-D

bobrippling, 2024-11-14:
Thanks - I believe the issue is in these lines of code:

https://github.com/espruino/BangleApps/blob/a1d140f5b800734d5c2d94f0e032e38065ef822e/apps/android/boot.js#L29-L30

and I suspect the full error message is `Uncaught Error: Module "messages" not found`

@girardcarillo, @signaleleven - do you have the [`messages` app](https://espruino.github.io/BangleApps/?id=messages) installed on your watch? And if so, what version does the apploader report it to be?

Could you both let me know what version of [Android Integration](https://espruino.github.io/BangleApps/?id=android) you have too?

girardcarillo, 2024-11-15:
So I didn't have this `Messages` app installed.
I just installed it (v 0.62), and tried to trigger again the bug.
Now the lines displayed don't talk anymore about the `Messages` module, but I still have something about Bluetooth connection (see screenshot).
[image]

And `Android Integration` is v 0.39

girardcarillo, 2024-11-15:
Yep, it's confirmed: as @signaleleven, Settings->Utils->Debug `off` and a Rewrite Settings solved the issue: no more messages on the screen when Bluetooth is turned on/off.

@bobrippling, do you know if this is a normal feature? Maybe having these messages on the main screen when Debug is `on` is intended somehow

bobrippling, 2024-11-17:
Ah yes - the debug info is intentional, docs below if you're interested, specifically the bit about bluetooth that explains our problem

https://github.com/espruino/BangleApps/blob/9421179620e775f42d36897fb759f7fb07116cd2/apps/setting/README.md?plain=1#L58-L62

So I think that's case closed, unless the issue with the messages module was also causing a problem?

signaleleven, 2024-11-17:
In my case the confusion was due to the fact that I expected "log" to only log to file and not on the screen, but the Readme explains it. For me, it can be closed, but I piggybacked on the issue so it's not my call.

-------- Original Message --------

girardcarillo, 2024-11-18:
Everything makes more sense now. I had missed this info, thanks for the doc @bobrippling 

Agreed, we can close this issue.
At least, if future users wonder why they have text appearing on their screen, they will know why ^^

closed 2024-11-18 by girardcarillo

bobrippling, 2024-11-18:
Cool, thanks for both of your inputs!