espruino/BangleApps #3665: [HAsensors] Only does battery life, no 'unique ID' set.

andiohn, 2024-11-19:
### Affected hardware version

Bangle 2

### Your firmware version

2v24

### The bug

This app is awesome, but perhaps: 
1. could we choose the sensors to export? Enable / Disable with 'amount of time between reports'?
2. Unique ID is not set, so home assistant isn't happy. Seems like it could be autogenerated or something

### Installed apps

hasensor.

bobrippling, 2024-11-19:
Fair suggestions for possibly future features, tagging @rigrig just for visibility

rigrig, 2024-11-19:
1. Adding sensors should be easy enough, which ones do you need?
    Not sure about adding settings, maybe we could pick a reasonable interval, and having a few unused sensor entities in Home Assistant doesn't seem like a problem.

2. The sensors are [dynamically created](https://www.home-assistant.io/integrations/http/#sensor) in Home Assistant, there is no way to give them a Unique ID.  
   (You can work around this in HA by using template sensors)

andiohn, 2024-11-19:
SENSORS:
I was thinking:
1. Last Heartrate (or even you could poll it every X minutes)
2. Steps 
3. Baro or other sensors, if they're already on and reporting?

Mainly it would be cool to report if another app is already waking to display something, like the heartrate one; baring that, it would be cool if we could also set it to report every X minutes. I love having data lol.

UNIQUE ID:
Ah, that's too bad. Is it possible to add a random string on first report or something? Or can we pass the unique_id flag as part of the call?

bobrippling, 2024-12-20:
@andiohn I've merged @rigrig's sensor improvement, I'll close this issue if you'd like to try out the new version

closed 2025-03-05 by gfwilliams