espruino/BangleApps #3666: [bangle.js gadgetbridge falling behind?] I had to install the other gadgetbridge along with ours to get my smart ring

andiohn, 2024-11-19:
### Affected hardware version

Bangle 2

### Your firmware version

2v24

### The bug

Is there a reason why we can't use the normal version? I know it had something to do with http intents? Is this still the case? Thanks. 

### Installed apps

gadgetbridge and bangle gadgetbridge

bobrippling, 2024-11-19:
Yes, I believe the normal version won't accept HTTP as a feature upstream. Also tagging @gfwilliams - I guess it's a bit behind upstream

thyttan, 2024-11-19:
As per https://www.espruino.com/Gadgetbridge you can install nighty builds from here https://freeyourgadget.codeberg.page/fdroid/repo/ . The "Bangle.js Gadgetbridge" nightly is up to speed with the regular Gadgetbridge.

andiohn, 2024-11-19:
Super cool! I had a thought that a plugin system would be the best of both worlds but I guess that this is the next best thing.

Thanks all!

andiohn, 2024-11-19:
Is there anyway to move from one to the other? The nightly doesn't share data between the normal one, that's the only downside I can see right now. I'll try to export or something.

thyttan, 2024-11-19:
Yes. Side bar - > Data management - > Export zip/ Import zip

That should cover everything.

gfwilliams, 2024-11-20:
Yes, upstream Gadgetbridge was super focussed on having *no* communications with the internet. I think even to the point where we couldn't have a WebView for the App Loader. While a plugin approach was suggested (having Gadgetbridge and *then* another app) it's not very user friendly, and given Gadgetbridge wasn't even on the App Store and the developers didn't have much interest in adding it, having this arrangement seemed best for all involved (I talked with Andreas about this first to make sure everyone was happy).

But the upshot is that I have to manually do updates to the play store app every so often, and i have a *lot* of other stuff to juggle so it doesn't happen all that often

I'll have a go at pushing an update soon, but I'm waiting for my change to activity reporting to get merged first: https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/4323

gfwilliams, 2025-03-05:
I'm pretty sure I did push a new Gadgetbridge a few weeks ago now

closed 2025-03-05 by gfwilliams