espruino/BangleApps #3667: [OpenHaystack (AirTag)] Module "ble_advert" not found

ZuriMuri, 2024-11-20:
### Affected hardware version

Bangle 2

### Your firmware version

2v24

### The bug

When I try to install 'OpenHaystack (AirTag)' with a valid advertisement key via the app loader on my Bangle 2 I get the following error: 

"Uncaught Error: Module "ble_advert" not found at line 23 col 21 in .boot0"

I suppose that the "ble_advert" package is not available anymore or at least faulty. My Console prompts this: 

Loading modules
loadModule(ble_advert)
ERROR: getURL("https://www.espruino.com/modules/ble_advert.min.js") error : HTTP 404
ERROR: getURL("https://www.espruino.com/modules/ble_advert.js") error : HTTP 404
ERROR: getURL("https://banglejs.com/apps/modules/ble_advert.min.js") error [object ProgressEvent]

I tried to manually add the module via "require("ble_advert")" and require("https://raw.githubusercontent.com/espruino/BangleApps/refs/heads/master/modules/ble_advert.js") within the IDE but it failed to RAM and FLASH. When trying to FLASH I needed to factory reset my bangle 2 because it always gets stuck on the Bluetooth loading screen.

[image]

bobrippling, 2024-11-21:
@gfwilliams we have the [`ble_advert` module](https://github.com/espruino/BangleApps/blob/master/modules/ble_advert.js), so I guess this is an issue with it being published to espruino.com?

gfwilliams, 2024-11-21:
Thanks for the report! 

> I tried ....  but it failed to RAM and FLASH

You need to upload it to Storage with the filename `ble_advert` and it'd be fine. There's some info about it in https://github.com/espruino/BangleApps/blob/master/modules/README.md

... but you shouldn't have to do that!

Looks like this would have started with bcc5ae9a155402b1d804264b7cb929bbc28898f3 which is actually one of your commits @bobrippling ? I doubt this has ever worked since that change.

The App Loader parses and handles modules for any file uploaded with a '.js' extension in the url - but there was no URL specified for the file so it just copied it in verbatim. I've just pushed a fix which supplies a URL which even though it's not used and wouldn't work anyway fools the app loader into parsing the file.

Just FYI, I'm not sure where you saw those `ERROR: getURL` messages, but the App Loader should be checking https://www.espruino.com/modules/ and https://banglejs.com/apps/modules for minified and unminified files. It errors for the first 3 and that's expected, but presumably it does find https://banglejs.com/apps/modules/ble_advert.js  and uploads it

ZuriMuri, 2024-11-21:
Thank you both for the fast replies and fixes!

> You need to upload it to Storage with the filename ble_advert and it'd be fine. There's some info about it in https://github.com/espruino/BangleApps/blob/master/modules/README.md

You are right - I saved the file with a ".js" extension before and that was my mistake. Uploading it manually made the errors go away.

> Just FYI, I'm not sure where you saw those ERROR: getURL messages, but the App Loader should be checking https://www.espruino.com/modules/ and https://banglejs.com/apps/modules for minified and [...]

The errors occured but if I recall correctly it then prompted that the minified file was loaded anyways.

Unfortunately I still can't get it to work... The key is not picked up by the apple network. I assume that the MAC address of my Bangle2 does not stay "static" as I see changing MAC addresses when I connect the app loader or IDE. The key I'm using is working with my FlipperZero over an [anisette-v3-server](https://github.com/Dadoum/anisette-v3-server) and [FindMy](https://github.com/biemster/FindMy) - I've tested that before. 
Maybe someone else can try if it works for them or if I need to make further adjustments to my Bangle settings for it to work?!

ZuriMuri, 2024-11-22:
There was indeed an error which I was able to solve. However the MAC does not stay the same with the Bangle but keeps changing after I upload the key.

I analyzed both payloads and the ble advertisement is identically (no error there). So I'd assume that the reason it is not working is that the bangle still changes its MAC eventhough it is not supposed to :(

ZuriMuri, 2024-11-22:
So I tried to go with a clean uninstall/install but still can't figure out why the signal is not picked up by the FindMy network. All it ever states is: "BLE connected, queuing BLE restart for
later". 

When your changes in the Git Repo are pushed to the official bangle app website I'll gladly give it another try...

In the meantime someone else can maybe test if it is working for them or if they are having similar issues?! 😕

gfwilliams, 2024-11-25:
> All it ever states is: "BLE connected, queuing BLE restart for later".

That's what you'd expect - it can't change some things while the BLE connection is active as it has to restart the bluetooth stack for them.

It's not possible you're just constantly staying connected to the Bangle so it's not getting a chance to advertise is it?

bobrippling, 2024-11-28:
To double-check, is this with the code revert to before bcc5ae9?

ZuriMuri, 2024-11-28:
I tried both actually, the NRF.setAdvertising and require("ble_advert") and they didn't work for me :/

bobrippling, 2024-11-29:
Interesting - have you seen it working in the past, ever? If not, it could be that your watch isn't being picked up by any devices capable of adding it to the FindMy network

bobrippling, 2024-11-29:
Similarly - I haven't been able to get my watch on the network for a while (#3566), perhaps something's up

ZuriMuri, 2024-11-30:
> Interesting - have you seen it working in the past, ever? If not, it could be that your watch isn't being picked up by any devices capable of adding it to the FindMy network

The key(s) - yes. If I add them to my FlipperZero they get picked up in an instance by the FindMy network. With the Bangle however they were never picked up.

Is there a way to troubleshoot the MAC address setting on the bangle? I checked BLE with a BLE scanner and the payload for advertising the key is the same as e.g. with the FlipperZero. It just seems that the MAC still keeps on changing.

bobrippling, 2024-11-30:
Yes, I see the same - I wonder if the advertisement has to be stable/unchanging for a certain duration before it'll be accepted? I've bumped the [advertising interval](https://www.espruino.com/ReferenceBANGLEJS2#l_NRF_setAdvertising) on mine and confirmed there's 5 seconds while a MAC is used. Will see if it gets picked up:

```diff
 {
 const key = E.toUint8Array(atob("...")); // public key
 const mac = [ key[0] | 0b11000000, key[1], key[2], key[3], key[4], key[5] ].map(x => x.toString(16).padStart(2, '0')).join(':'); // mac address
 const adv = [ 0x1e, 0xff, 0x4c, 0x00, 0x12, 0x19, 0x00, key[6], key[7], key[8], key[9], key[10], key[11], key[12], key[13], key[14], key[15], key[16], key[17], key[18], key[19], key[20], key[21], key[22], key[23], key[24], key[25], key[26], key[27], key[0] >> 6, 0x00 ]; // advertising packet
 NRF.setAddress(mac);
 //require("ble_advert").push(adv, {whenConnected: true, interval: 1000}); // advertise AirTag *and* normal device name (to remain connectable)
-NRF.setAdvertising([adv,{}],{whenConnected: true, interval: 1000}); // advertise AirTag *and* normal device name (to remain connectable)
+NRF.setAdvertising([adv,{}],{whenConnected: true, interval: 5000}); // advertise AirTag *and* normal device name (to remain connectable)
 }
```

bobrippling, 2024-12-01:
No reports - I suspect the issue isn't the `interval`. Given there's no difference between this and the code when I saw it working (in 008fce39f026a774), I wonder if there's other changes that could be causing it.

Do you see any difference in the BLE reports from your FlipperZero vs the Bangle? I see the advertisements parsed as FindMy reports in NRF Connect, with the public key payload matching my public key

ZuriMuri, 2024-12-02:
Here is what the FlipperZero advertises

> Device Name: C6-XX-YY-ZZ-85-C1
> Device MAC: C6:XX:YY:ZZ:85:C1
> First seen: Mon Dec 2 09:37:00 2024
> Last seen: Mon Dec 2 09:40:44 2024
> Advertising interval: 8939 ms
> Advertisement packet(s):
> MFD > 09:37:11.146 > 0x004C (Apple, Inc.) > (27 bytes) 0x1219a0RESTOFTHEKEYed0100

and here the Bangle

>Device Name: Bangle.js a14b
>Device MAC: 74:XX:YY:ZZ:2E:6D
>First seen: Mon Dec 2 09:39:47 2024
>Last seen: Mon Dec 2 09:43:37 2024
>Service(s) advertised:
>{6E400001-MYSE-RVIC-EADV-ERTE24DCCA9E}
>Advertising interval: 7192 ms
>Advertisement packet(s):
>MFD > 09:39:52.515 > 0x004C (Apple, Inc.) > (27 bytes) 0x121900RESTOFTHEKEYed0100

The differences I see:

1. Device name and MAC are the same (and always stay the same) with the FlipperZero
2. Bangle advertises 0x121900... whilst FlipperZero advertises 0x1219a0... but if I remember correctly I read somewhere that this is only distinguishing between one being "I am an Apple device" and the other "I am no Apple device"

The broadcast intervall on the Flipper can be manually set anywhere between 1-10 sec. Other than that the keys are identitical.
So it must have to do something with the device name and/or MAC that is being "ignored" by the FindMy network.

gfwilliams, 2024-12-02:
Do you think one of you could send me a key from OpenHaystack to test? privately if you want.

I'm pretty sure the interval shouldn't have any effect - once the advertising is set it'll stay as is.

I just tested on a Bangle here for the MAC address - I uploaded `NRF.setAddress("ff:ee:dd:cc:bb:aa");` via the Web IDE and disconnected, and I scanned with NRF connect, and I see exactly that - the Bangle always stays advertising with a MAC address of ff:ee:dd:cc:bb:aa and it doesn't change.

But you're saying that yours does change - what kind of addresses does the MAC address change between?

Maybe you could try uploading a file called: `debug.0.boot.js` with the Web IDE

```
NRF._setAddress = NRF.setAddress;
NRF.setAddress = function(addr) {
  console.log("Address changed to "+E.toJS(addr));
  NRF._setAddress(addr);
  throw new Error("Address changed to "+E.toJS(addr));
};
```

Then reload and see what's printed. Maybe also try with the Open Haystack app removed.

It will throw an exception so it may break what's running, but it will give you a stack trace of where setAddress was called from

Interesting about the A0/00 - that's a trivial change to make and might help I guess?

gfwilliams, 2024-12-02:
It's hard to be 100% sure, but the line `NRF.setAddress(mac);` exactly matches what's in OpenHaystack, which makes sense. Please can you try changing the boot code to:

```
NRF._setAddress = NRF.setAddress;
NRF.setAddress = function(addr) {
  console.log("Address changed to "+E.toJS(addr));
  NRF._setAddress(addr);
  if (addr!="c6:a4:fc:6f:85:c1") throw new Error("Address changed to "+E.toJS(addr));
};
```

So it'll run openhaystack without issues - then leave the bangle for a bit and play around, and see if it gets called any other time with anything else?

> I'll pass a new key via direct E-Mail to you!

Thanks!

ZuriMuri, 2024-12-04:
> So it'll run openhaystack without issues - then leave the bangle for a bit and play around, and see if it gets called any other time with anything else?

It did not throw an error or get called any other time with anything else.

I've noticed that when I am connected via browser with the espruino ide - I see the following (when trying to connect in a different tab to the app store e.g.):
[image]

When the Bangle is disconnected and I try to re-connect I get:
[image]

Does that mean that the "service" is advertised differently and because of the `NRF.setAddress` it doesn't show any MAC for the connection? Or am I just mixing things up?

bobrippling, 2024-12-04:
I see exactly the same - different devices show up, sometimes unidentified rather than `Bangle.js <id>`, which led me to think the mac was cycling

@gfwilliams let me know if you want another public key to test with and I'll whizz mine over

> Interesting about the A0/00 - that's a trivial change to make and might help I guess?

This could be a point

ZuriMuri, 2024-12-04:
> Bangle advertises 0x121900... whilst FlipperZero advertises 0x1219a0... but if I remember correctly I read somewhere that this is only distinguishing between one being "I am an Apple device" and the other "I am no Apple device"

I've found the [website ](https://adamcatley.com/AirTag.html#advertising-data) but the byte A0/00 is part of the "Status byte" and not the "company identifier" if I interpret it correctly. So this change might really be worth a try.

gfwilliams, 2024-12-04:
> I've noticed that when I am connected via browser with the espruino ide - I see the following (when trying to connect in a different tab to the app store e.g.):
> ![Screenshot 2024-12-04 073900](https://private-user-images.githubusercontent.com/113093915/392260431-8105c0ea-31a3-4787-b5b0-3f859d942732.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMzMDI2NDgsIm5iZiI6MTczMzMwMjM0OCwicGF0aCI6Ii8xMTMwOTM5MTUvMzkyMjYwNDMxLTgxMDVjMGVhLTMxYTMtNDc4Ny1iNWIwLTNmODU5ZDk0MjczMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwNFQwODUyMjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNjU0ZTcwYTE0YTgyYzk4NTU1MmJiY2I2Mjk3NzkxODdmMDZlZTViOTdkMjcxOTRjZTFmNjAzZjZkMWRhYjlhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.h1hBPMkuY0fbesjIaMigAQDrzNcajmZ2LvCl4_E9xk4)

But is it just two addresses? Or more than two?

Because that's exactly what you would expect - the Bangle's own address from before the MAC address is set (which may happen for a fraction of a second at boot or before you long-press to reload apps after installing OpenHaystack) and then the new address you set.

* One should have the last 4 digits `ab:cd` matching `Bangle abcd`
* One should match whatever that code I posted up said in the console

It's really not helping to debug this when you appear to be changing keys between posting up and also blanking out your full key and MAC address. But I used the key you gave and what I see here is...

I pasted in the `Advertising key`, and:

* The MAC address the Bangle has after restarting matches what NRF.setAddress said, which is the first 6 bytes of the key (with the first one ORed slightly) as is done on https://github.com/seemoo-lab/openhaystack/blob/main/Firmware/Microbit_v1/offline-finding/main.c and https://github.com/seemoo-lab/openhaystack/blob/main/Firmware/ESP32/main/openhaystack_main.c#L107
* According to NRF Connect it's not changing MAC address from what it was asked to use
* According to NRF connect the Public key being advertised exactly matches what I pasted in
(you can use `print(E.toUint8Array(atob("base64_key")).slice().map(x=>x.toString(16).padStart(2,0)).join(""))` to change your key from b64 to hex, and chars 2..11 should match the MAC address too)

So it appears to all be doing the correct thing as far as I can tell. And both the example bits of code I mentioned for OpenHaystack report 0x121900 too.

So honestly I'm not really sure what the difference would be, unless Apple have now started filtering devices that also advertise other information (like their name) as well as the AirTag packet.

bobrippling, 2024-12-11:
> I've found the [website ](https://adamcatley.com/AirTag.html#advertising-data) but the byte A0/00 is part of the "Status byte" and not the "company identifier" if I interpret it correctly. So this change might really be worth a try.

I suspect that'll be related to my PR here #3566

> So it appears to all be doing the correct thing as far as I can tell. And both the example bits of code I mentioned for OpenHaystack report 0x121900 too.
>
> So honestly I'm not really sure what the [...]

I suspect this is the case - no hits for me despite using code where I had gotten hits several months back