espruino/BangleApps #3673: Error in settings!

voloved, 2024-11-27:
When I click `Settings -> Apps -> Messages`, it displays:

```
Messages:
Error in
settings!

BTN1 to go
back
```

with no further context.

Septolum, 2024-11-27:
@voloved what version of messages do you have installed? That PR was made to fix the error you're seeing (or one similar), and merged as version 0.65

voloved, 2024-11-27:
> @voloved what version of messages do you have installed? That PR was made to fix the error you're seeing (or one similar), and merged as version 0.65

It looks like 0.64 is the default on Bangle.js's app loader. Also I backed up the watch and saw `!!settings().ignoreUnread,` in `messages.settings.js` in the zip.

[image]

voloved, 2024-11-27:
> Interesting - do you see anything if you do exactly the same, but with [the IDE](https://www.espruino.com/ide/) connected? Any messages in the IDE's console?

Here's my output:

```
      58
Press  96 7 38
Messages settings error: SyntaxError: SyntaxError: Got ';' expected ','
```

Note:  I did manually modify the `messages.settings.js` by backup, edit the file, and restore to try to fix this issue, so that comma may be a typo?

voloved, 2024-11-27:
Fixed.

1.  in `https://www.espruino.com/ide/#` connect to the watch.
2. Access device storage and `Save` `message.settings.js`
3. Open the saved file in `Notepad++` (`VSCode` gave me issues, likely due to odd saving on encoded symbols it can't read)
4. Delete `()` in `settings().ignore...` and save.
5. At the top of the same Device Storage window in the IDE that opened the file, click upload.
6. Upload the saved file.
7. Click `Run File` next tp the the newly added file to make sure it can be read properly.

I'll expect v0.65 to not have this issue.

bobrippling, 2024-11-28:
> It looks like 0.64 is the default on Bangle.js's app loader

Yes, there's a delay before apps go onto the app loader - the dev one has the latest versions immediately, so it's [there already](https://espruino.github.io/BangleApps/?id=messages).

Thanks for the details too - I'll close this for now but if it reappears let us know

closed 2024-11-28 by bobrippling