espruino/BangleApps #3694: [gpssetup] PSMOO setting not working as intended

nravanelli, 2024-12-12:
### Affected hardware version

Bangle 2

### Your firmware version

2v24

### The bug

Unsure if this is a bug, but seems to not work as intended/explained. 

My assumption with the PSMOO option, using default settings results in the GPS module searching for 5-6 seconds and then only attempting to find a GPS connection every 2 minutes (or 120 seconds).

However, after attempting this on two watches, the eventlistener for `Bangle.on('GPS');` continues to respond, and looks like GPS coordinates are continuing to update, even after a fix is obtained. Is this expected ? The suggested battery life when running in PSMOO is also not what I observed (<10 h from 95% charge), which leads me to the conclusion that this isnt expected.

Code to replicate is below:

      function logGPS(f){
          console.log(f);
      };
      Bangle.on('GPS', logGPS);
      Bangle.setGPSPower(1);
      require("gpssetup").setPowerMode({power_mode:"PSMOO"})
        .then(function() {Bangle.setGPSPower(1);
        });

gfwilliams, 2024-12-13:
The gpssetup app is only listed as for Bangle.js 1, not Bangle.js 2. When 'Bangle.js 2' is listed as the watch type in the App Loader it shouldn't even be visible to install?

It'll be sending config commands meant for the wrong GPS chip to the Bangle so it won't have any effect at all I'm afraid.

nravanelli, 2024-12-13:
Read the documents, one must... 

I'll revise and commit a modified gpssetup module to include version checking for BangleJS2. with setIntervals/timeouts to emulate PSMOO on the BangleJS1

gfwilliams, 2024-12-13:
Great, thanks! Maybe the app loader should be more insistent about installing apps that aren't meant to work...

gfwilliams, 2025-03-05:
Looks like this is fixed now by @nravanelli's PR

closed 2025-03-05 by gfwilliams