espruino/BangleApps #3707: [calendar][agenda] Feature request: calendar view of Gadgetbridge-synchronized events

randyheydon, 2025-01-05:
`calendar` is a great app, having a nice calendar view and a nice agenda view.  But it only shows events that have been explicitly added to the app; it does not show any events from my regular calendar on my phone (it appears to read from "calendar.json").

`agenda` is a great app, having a nice agenda view of events synchronized from my phone via Gadgetbridge (it appears to read from "android.calendar.json"), and a detail view for each event.  But it can't show those events in or alongside a calendar view.

I'm using both apps, but I find myself switching back and forth between them to make sense of when things are happening.  It would be nice to have one app that can show synchronized events in a calendar view.  I see three possible approaches for this:
1. Enhance `calendar` to also read events from the synchronized calendar.
2. Enhance `agenda` to add a calendar view.
3. Create a new app that does all of this.

For reference, I have a Bangle 2 on firmware 2v25, with `calendar` 0.18 and `agenda` 0.15.  I also gave a quick try to `rebbleagenda` 0.02 and `calclock` 0.09, but like `agenda`, those do not have calendar views.

nxdefiant, 2025-01-07:
I would prefer option 4. Have calendar and agenda use the same appointment source and only this. I see no point in having multiple apps using their own data source. This should be handled similar to waypoints or scheduler.

I was not aware of agenda when I added the holidays to the simple calendar, this is why it uses its own. Agenda seems to read android.calendar.json - does anyone have a description of this file? Does it cover all features of calendar? Basically it is only
- day
- title
- type holiday/other
- repeat yearly

nxdefiant, 2025-01-07:
> The "type" entries aren't the same between calendar.days.json and android.calendar.json (int vs. string)

as long as we can identify a mapping this should be ok

> there's no repeat information at all in android.calendar.json

can we just add a field? Non Gadgetbridge users will just ignore it and it just won't be set or needed for Gadgetbridge user.

> Also, I suspect that android.calendar.json is effectively read-only, since it's synced from Gadgetbridge. It might be possible to write to android.calendar.json, but those writes won't be sent back to the phone and may be overwritten by Gadgetbridge (I'm guessing, this _might_ be avoidable if id values don't conflict). If you want `calendar` to drop use of calendar.days.json entirely, then it might have to give up the ability to add entries, making it unable to work well without a paired phone. That wouldn't bother me, but other users might not like that.

This is not good. However see answer to point 5 below

> 1. Switch the source being read from.

Sure, including a one time migration in app or via interface.html

> 2. Adapt the timestamp entry into a Date object (including start time).

Should not be a problem

> 3. Do something to guess whether or not its a holiday (maybe using allDay?).

I thought this is what type is for? What are the possible values for type? And no allDay is not enough: "other" and "holiday" are both all day.

> 4. Add something to display extra data like description, location, durationInSeconds, calName.
 
Like sched/alarm calendar currently uses only the title. IMHO having longer texts on the small display is not advantageous.

> 5. Remove or adapt ability to add/edit holidays.

We could only allow to modify dates from calendar if gadgetbridge is not installed.

> These changes seem pretty significant to me, but @nxdefiant, you know `calendar` better than I do. Still think it's the best approach?

Honestly I hate calendar.days.json because calendar is the only user of this file. But yes it would be much simplier to just read android.calendar.json and append its data to the event array in https://github.com/espruino/BangleApps/blob/master/apps/calendar/calendar.js#L62

randyheydon, 2025-01-08:
> > 3. Do something to guess whether or not its a holiday (maybe using allDay?).

> I thought this is what type is for? What are the possible values for type?

I did some digging in Gadgetbridge, and it seems to use type [only for sunrise and sunset](https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/model/CalendarEventSpec.java#L23-L25) (and that support [isn't even enabled for Bangle.js](https://codeberg.org/Freeyourgadget/Gadgetbridge/src/branch/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L1840)).  I don't see anything that maps neatly to `calendar`'s holiday/other type.  As you suggested for repeat, perhaps that could be an extra field.  But I'm new here, I can't say how well extra fields would work.

> Like sched/alarm calendar currently uses only the title. IMHO having longer texts on the small display is not advantageous.

`agenda` does a decent job of conveying this information, with date and location in small text (see screenshot below), and all info in a scrollable window when an agenda item is selected.  That extra information is often important, so I'd like some way to see it (ideally without going to another app).  But I certainly won't dictate what the UI should be for that :).

[image]

nxdefiant, 2025-01-08:
ok this holiday thing is definitely an issue. What is this :birthday: in your screenshot? Is that just text in the title?

randyheydon, 2025-01-08:
It's a birthday cake emoji in the title, but my Bangle renders all emojis as text surrounded by colons (including in other apps like `messages`).