espruino/BangleApps #3708: [hadash] error: TBD - not correctly configured!

BenJamesAndo, 2025-01-06:
Hey @flaparoo thanks for making this app. I'm super keen to try it out. However I can't get it to work. No matter what config I do I always get undefined - `not correctly configured!`, `TBD - not correctly configured!` or `Fixed - not correctly configured!`

[image]
I've tried using some of the example code such as 
```
{
  "type": "service",
  "title": "Custom Notification",
  "domain": "persistent_notification",
  "service": "create",
  "data": {
    "title": "Fixed"
  },
  "input": {
    "message": {
      "options": [],
      "value": "Pre-filled text"
    },
    "notification_id": {
      "options": [
        "123",
        "456",
        "136"
      ],
      "value": "999",
      "label": "ID"
    }
  }
}
```
but I get the same error in the app. I have Allow Intents and Allow internet Access in Bangle.js Gadgetbridge. If I incorrectly enter in the Long Live Token then I don't even get any errors in HA, so I think it's just to do with the config as it's not even getting far enough to hit my HA server. I've tried with HTTPS and HTTP version of HA URL with /api at the end. I'm running the latest firmware and app with a Bangle.js 2.

Any help would be greatly appreciated! Thanks.

flaparoo, 2025-01-06:
Hi @BenJamesAndo 
The error messages are generated when the menu configuration is not correct. Please make sure the list of menu entries is in a JSON array, even if you only have 1 entry. This is an example of the complete JSON code for 2 menu entries:
```
[
  {
    "type": "state",
    "title": "Check for updates",
    "id": "update.home_assistant_core_update"
  },
  {
    "type": "service",
    "title": "Create Notification",
    "domain": "persistent_notification",
    "service": "create",
    "data": {
      "message": "test notification",
      "title": "Test"
    }
  }
]
```
Note the square brackets around both entries.

If that doesn't solve the issue, please post your complete menu config (I suggest to switch to the "text" view in the JSON editor, and disable the indentation/formatting).

BenJamesAndo, 2025-01-06:
Hey @flaparoo that's now solved the buttons, they're now displaying correctly. However now I'm getting `Error querying state!`
[image]
No logs in HA.

BenJamesAndo, 2025-01-06:
I got it working. It has to use a https url. I think I read somewhere that the Banglejs app doesn't work with http only https. Thanks for your help with the JSON.

closed 2025-01-06 by BenJamesAndo