espruino/BangleApps #3718: [app/widget name] Short description of bug

Sleuth56, 2025-01-19:
### Affected hardware version

Bangle 2

### Your firmware version

2v25

### The bug

Over night the watch will become disconnected from gadgetbridge. This happens every night. To fix it you have to reload the watch. You don't have to watchdog it. 

### Installed apps

qrcode (0.06)
pebblepp (0.10)
messageicons (0.08)
reply (0.03)
launch (0.21)
locale (0.20)
icons (0.02)
widalarm (0.02)
alarm (0.50)
boot (0.65)
clock_info (0.14)
clockbg (0.06)
widmessages (0.07)
android (0.39)
clkinfocal (0.05)
clkinfomsg (0.01)
sched (0.30)
2047pp (0.04)
touchtimer (0.06)
qmsched (0.10)
setting (0.75)
smpltmr (0.11)
messages (0.65)
messagegui (0.84)
magnav (0.07)
hidcam (0.04)
tetris (0.12)

Sleuth56, 2025-01-19:
#3717

thyttan, 2025-01-21:
Would you be comfortable with making a backup-zip from the app loader and sjare it with me (beware if you have sensitive info on the watch)?

If so I could restore it to one of my watches and see if it reproduces on my end.

Sleuth56, 2025-01-23:
I removed the health and calendar data. Doesn't look like there is any other personal data. So here's hoping, 😆 . Those shouldn't change the functionality I don't think.
[Banglejs backup.zip](https://github.com/user-attachments/files/18514272/Banglejs.backup.zip)

thyttan, 2025-01-28:
Tried it one night now. No crash. I'll let it stay connected to see if it happens eventually.

thyttan, 2025-01-29:
Been connected now for some days without crashing or disconnecting.

Do you have some specific usage pattern that could be the difference for us? 🤔

Sleuth56, 2025-01-31:
Other than sending navigation information I don't think I have anything strange. I've never gotten the nav stuff working. I just turned it off on my end so we will see. It's been better recently. Not sure why, I didn't change anything.

thyttan, 2025-02-04:
The watch eventually ran the battery flat. Before that it seemingly never experienced the described bug.

@Sleuth56 how's the last couple of days been for your watch?

Sleuth56, 2025-02-04:
It's still doing it. I woke up this morning and the step counter was still what it was yesterday. Meaning that the midnight reset function hadn't triggered. Resetting the watch reset the steps to 0.

thyttan, 2025-02-04:
Have you enabled logging via "Settings"->"Utils"->"Debug"->Select "Log" ? That will generate a "log.txt" file that can be looked at after the fact to get a clue of what goes wrong.

Sleuth56, 2025-02-07:
I haven't but I will do that

thyttan, 2025-02-16:
Hi again @Sleuth56! Did you ever find out what the problem was with the logging turned on?

Sleuth56, 2025-02-16:
No I didn't. I turned the logging on and even caught it right as it happened but the log file didn't have anything other than a few Bluetooth lines that were just status changes. You know connected and disconnected.

thyttan, 2025-02-16:
To add to this report,

1. Does the watch freeze and become unresponsive before you reload it - or is it just that it disconnects from the phone and Gadgetbridge?
2. What flavor (Bangle.js or regular) and version number of Gadgetbridge do you use?
3. What model phone and android version do you use?
4. Have you exempt Gadgetbridge from battery optimization on the watch - letting it run unrestricted in the background?

gfwilliams, 2025-03-04:
Just to add to this - if battery optimisations haven't been turned off for Gadgetbridge then it may well be killed by the OS around midnight.

Gadgetbridge also may not automatically reconnect if it loses connection - there is an option in Gadgetbridge for that.

If you've got some other device that is paired to the Bangle like a PC then it's possible that the PC could be jumping in connecting as soon as Gadgetbridge disconnects?

gfwilliams, 2025-03-26:
Closing due to lack of replies

closed 2025-03-26 by gfwilliams