espruino/BangleApps #3728: [clock_info] Single clock_info throws error if not in the Bangle category

WhelanB, 2025-02-01:
### Affected hardware version

Bangle 2

### Your firmware version

2v24

### The bug

### Description
When running a custom clock info, the object returned by the clock info has a name and an items array containing individual clock info objects. If the name of the object is not "Bangle" (as the clock_info module provides default clock_infos under the name "Bangle", so the number of clock_infos is never 1), and the items array is of length 1, attempting to use the clock info twice on the same clock (Pebble++, but also tested Slope Clock) results in a hang and the following error:

```javascript
Uncaught Error: Can't read property 'on' of undefined
 at line 12 col 3447 in clock_info
...andler=()=>drawItem(itm);itm.on("redraw",options.redrawHandle...
                               ^
in function "menuShowItem" called from line 12 col 4518 in clock_info
...menuA].items[options.menuB]);}E.stopEventPropagation&&E.stopE...
                               ^
in function called from system
```

### Expected behaviour:
The clock info should appear twice on the same clockface

### Steps to reproduce:
- Add the following default clock_info (Timer from the BangleJS docs) to your watch as a clock_info:
```javascript
(function() {
  var startTime;
  var interval;
  var running = false;

  return {
    name: "OUTER_NAME",
    items: [
      { name : "INNER_NAME",
        get : () => ({ text : startTime ? ((Date.now()-startTime)/1000).toFixed(1) : "--",
                      img : atob("GBiBAAAAAAB+AAB+AAAAAAB+AAH/sAOB8AcA4A4YcAwYMBgYGBgYGBg8GBg8GBgYGBgAGAwAMA4AcAcA4AOBwAH/gAB+AAAAAAAAAA==") }),
        show : function() { // shown - if running, start animation
          if (running)
            interval = setInterval(()=>this.emit('redraw'), 100);
        },
        hide : function() { // hidden - stop animation
          if (interval) clearInterval(interval);
          interval=undefined;
        },
        run : function() { // tapped - cycle between start and stop
          if (interval) { // stop
            clearInterval(interval);
            interval=undefined;
            running = false;
          } else { // start
            interval = setInterval(()=>this.emit('redraw'), 100);
            startTime = Date.now();
            running = true;
          }
        }
      }
    ]
  };
});
```
- Using a clockface such as Pebble++, try to set both clock_info areas to this clock info.
- The clock will become unresponsive until it is reset, and throw an error visible in the IDE
- Change the ``OUTER_NAME`` string to ``Bangle`` and reinstall the clock_info
- Try to set both clock_info areas to this clock_info
- Observe that the clock continues to function as normal.

bobrippling, 2025-02-02:
Sounds like a problem with overwriting built-in clkinfos - `.items[options.menuB]` here is `undefined`:

https://github.com/espruino/BangleApps/blob/2954d513b8e060839fbd066c914e1bc45f0efdc8/apps/clock_info/lib.js#L293

likely because there was existing config for the old bangle menu items - the menu items have changes and so the old config is now invalid. @gfwilliams is this intended to be overwritten? If so, I expect we'll want to validate the config, if not the bug is in the user code

gfwilliams, 2025-02-03:
Thanks for the report! From what I remember menuA and menuB are always supposed to point to something. I'm pretty sure the issue is:

https://github.com/espruino/BangleApps/blob/2954d513b8e060839fbd066c914e1bc45f0efdc8/apps/clock_info/lib.js#L285-L288

> "When we change, ensure we don't display the same thing as another clockinfo if we can avoid it"

but the length check is off by 1. In this case length=1 but index=0, so `0<1 = true `and it thinks it can increment `menuB`

I'll test and put a fix in now

gfwilliams, 2025-02-03:
Yes - all sorted. It'll be fixed in the development app loader

bobrippling, 2025-02-03:
Nice!

WhelanB, 2025-02-05:
Great, thanks for the quick fix!

commit 0d3cad3d (2025-02-03): 0.15: Fix error when displaying a category with only one clockinfo (fix #3728)
commit 62d3f803 (2025-02-04): 0.15: Fix error when displaying a category with only one clockinfo (fix #3728)