espruino/BangleApps #3729: Changing Default Branch to Build Launcher After Forking

voloved, 2025-02-02:
Hello,

Is it possible to change the default branch that the launcher builds off and sends you to via the source code button?

I tried:
1. Changing the Build Source in the Github Page settings,
2. Changing the branch name for `Const.APP_SOURCECODE_URL` in `loader.js`

My goal is for `master` to be a copy of the official repo for easy syncing, while a new `dev` repo is what builds the site.

Thank you

thyttan, 2025-02-02:
Just for reference.

 I use a branch different than "master" for building my app loader, like you say you do. I get all the apps and changes from that branch on the app loader. But the github octocat sourcecode button still takes me to my master branch.

Don't know how to change that though.

voloved, 2025-02-04:
Thanks! I tried it again today, and it worked! Likely a stale cookie issue.
> Don't know how to change that take you to the actually used branch though.

This change fixed that issue for me: https://github.com/voloved/BangleApps/commit/f64405399b11c042bfda888bda14149a7aa6add0

thyttan, 2025-02-04:
> This change fixed that issue for me: https://github.com/voloved/BangleApps/commit/f64405399b11c042bfda888bda14149a7aa6add0

Neat, thanks for sharing!

closed 2025-02-04 by voloved