espruino/BangleApps #3733: Removing pointless warnings?

gfwilliams, 2025-02-04:
Right now the sanity checker looks like this:

```
nown warning : App carcrazy has a setting file but no corresponding data entry (add `"data":[{"name":"carcrazy.settings.json"}]`)
Known warning : App gpsrec data file wildcard .gpsrc? does not include app ID
Known warning : App loadingscreen has a setting file but no corresponding data entry (add `"data":[{"name":"loadingscreen.settings.json"}]`)
::warning file=apps/messagesoverlay/settings.js::App messagesoverlay has a setting file but no corresponding data entry (add `"data":[{"name":"messagesoverlay.settings.json"}]`)
Known warning : App trex has a setting file but no corresponding data entry (add `"data":[{"name":"trex.settings.json"}]`)
Known warning : widhwt isn't an app (widget) but has an app.js file (widhwtapp.js)
Known warning : App owmweather data file weather.json is also listed as data file for app weather
Known warning : In locale it_CH, long time format might not work in some apps if it is not "%HH:%MM:%SS"
Known warning : In locale it_IT, long time format might not work in some apps if it is not "%HH:%MM:%SS"
Known warning : In locale wae_CH, long time format might not work in some apps if it is not "%HH:%MM:%SS"
Known warning : In locale wae_CH, short time format might not work in some apps if it is not "%HH:%MM"
Known warning : In locale test, long time format might not work in some apps if it is not "%HH:%MM:%SS"
Known warning : In locale test, short time format might not work in some apps if it is not "%HH:%MM"
Known error : In locale en_CA, long date output must be shorter than 15 characters (Wednesday, September 10, 2024 -> 29)
Known error : In locale fr_FR, long date output must be shorter than 15 characters (10 septembre 2024 -> 17)
Known error : In locale fr_FR, short month must be shorter than 5 characters
Known error : In locale sv_SE, speed must be shorter than 5 characters
Known error : In locale en_SE, long date output must be shorter than 15 characters (September 10 2024 -> 17)
Known error : In locale en_NZ, long date output must be shorter than 15 characters (Wednesday, September 10, 2024 -> 29)
Known error : In locale en_AU, long date output must be shorter than 15 characters (Wednesday, September 10, 2024 -> 29)
Known error : In locale de_AT, long date output must be shorter than 15 characters (Donnerstag, 10. September 2024 -> 30)
Known error : In locale en_IL, long date output must be shorter than 15 characters (Wednesday, September 10, 2024 -> 29)
Known error : In locale es_ES, long date output must be shorter than 15 characters (miércoles, 10 de septiembre de 2024 -> 35)
Known error : In locale fr_BE, long date output must be shorter than 15 characters (dimanche septembre 10 2024 -> 26)
Known error : In locale fr_BE, short month must be shorter than 5 characters
Known error : In locale fr_BE, short month must be shorter than 5 characters
Known error : In locale fr_BE, short month must be shorter than 5 characters
Known error : In locale fr_BE, short month must be shorter than 5 characters
Known error : In locale fr_BE, short month must be shorter than 5 characters
Known error : In locale fi_FI, long date output must be shorter than 15 characters (keskiviikkona 10. maaliskuuta 2024 -> 34)
Known error : In locale fi_FI, short month must be shorter than 5 characters
Known error : In locale fi_FI, short month must be shorter than 5 characters
Known error : In locale fi_FI, short month must be shorter than 5 characters
Known error : In locale fi_FI, short month must be shorter than 5 characters
Known error : In locale fi_FI, short month must be shorter than 5 characters
Known error : In locale fi_FI, short month must be shorter than 5 characters
Known error : In locale fi_FI, short month must be shorter than 5 characters
Known error : In locale fi_FI, short month must be shorter than 5 characters
Known error : In locale fi_FI, short month must be shorter than 5 characters
Known error : In locale fi_FI, short month must be shorter than 5 characters
Known error : In locale fi_FI, short month must be shorter than 5 characters
Known error : In locale de_CH, meridian must be shorter than 4 characters
Known error : In locale de_CH, meridian must be shorter than 4 characters
Known error : In locale de_CH, long date output must be shorter than 15 characters (Donnerstag, 10. September 2024 -> 30)
Known error : In locale fr_CH, long date output must be shorter than 15 characters (dimanche 10 septembre 2024 -> 26)
Known error : In locale fr_CH, short month must be shorter than 5 characters
Known error : In locale fr_CH, short month must be shorter than 5 characters
Known error : In locale fr_CH, short month must be shorter than 5 characters
Known error : In locale fr_CH, short month must be shorter than 5 characters
Known error : In locale fr_CH, short month must be shorter than 5 characters
Known error : In locale wae_CH, long date output must be shorter than 15 characters (Sunntag, 10. Herbštmánet 2024 -> 29)
Known error : In locale tr_TR, long date output must be shorter than 15 characters (10 Haziran 2024 Pazartesi -> 25)
Known error : In locale hu_HU, long date output must be shorter than 15 characters (2024 Szep 10, Csütörtök -> 23)
Known error : In locale oc_FR, long date output must be shorter than 15 characters (divendres 10 setembre de 2024 -> 29)
Known error : In locale oc_FR, short month must be shorter than 5 characters
Known error : In locale oc_FR, short month must be shorter than 5 characters
Known error : In locale hr_HR, meridian must be shorter than 4 characters
Known error : In locale hr_HR, meridian must be shorter than 4 characters
Known error : In locale hr_HR, short month must be shorter than 5 characters
Known error : In locale sl_SI, meridian must be shorter than 4 characters
Known error : In locale sl_SI, meridian must be shorter than 4 characters
Known error : In locale ca_ES, long date output must be shorter than 15 characters (10 setembre 2024 -> 16)
Known error : In locale ca_ES, short month must be shorter than 5 characters
```

The warnings were added ~4 months ago but I don't think they're likely to be fixed by anyone at any point, and they just get in the way of seeing real issues. Shall I disable these checks?

thyttan, 2025-02-04:
Could they be hidden under a drop-down arrow?

Either way, no strong feelings on my part. As you indicate, getting rid of them in the output would make it easier to spot more relevant problems. Maybe especially for new contributors.

gfwilliams, 2025-02-04:
Maybe I'll make it so you need to run `sanitycheck --show-all` to show known warnings for now?

thyttan, 2025-02-04:
I like the sound of that!

bobrippling, 2025-02-04:
Yeah me too, sounds good 👍

atjn, 2025-02-05:
Gating known warnings behind a flag sounds like a fine idea regardless, but I do have plans to fix most of these warnings.

closed 2025-02-05 by gfwilliams

gfwilliams, 2025-02-05:
Thanks! Just pushed a change (also gives a bit more info on known warnings when you do show-all) - it'd be great if you did get around to fixing some of those at some point (I have just fixed a bunch of the more basic ones).

commit 27bb244d (2025-02-05): remove known warnings from list by default so we can see new errors more clearly (fix #3733)