espruino/BangleApps #3741: Weather Clock wind speed in km/h?

stanelie, 2025-02-10:
Hello.
I probably can't read, but I am unable to change the wind speed display of the Weather Clock to KM/h instead of mph.

Can anyone point me to the right setting?
Thanks!

stanelie, 2025-02-11:
Nevermind.

Stupid me expected the setting to change the watchface display to be in the watchface settings.

Instead, I found an obscure thread on the internet revealing the need to change the locale using a separate app from the app store (the "languages" app). This fixed my issue.

It would have been helpful to have the setting in the correct place, in the watchface settings, or at the very least a button explaining that this feature could be adjusted using the Languages app.

Not intuitive.

thyttan, 2025-02-11:
Thanks for the suggestion! If you feel like it a PR with clarifications in the apps readme would be much appreciated 😸

thyttan, 2025-02-11:
Yeah, you could go about it in different ways. Easiest if you want something like that may be a shortcut from this app's settings page to just open the locale app settings page. Maybe that is clearer in regards go what happens behind the scenes? 

But I guess you could go with your idea as well - I'd merge it 👍

gfwilliams, 2025-03-05:
I've just added a message to the README for weatherclock and weather

The problem is that we have other apps that display weather - and it's not reasonable to expect all of those to have their own options for setting speed, which you'd have to change in every app. I think it's a *good* thing to have a way to set the units for speed globally - it just seems in your case it wasn't obvious where/how you had to do that.