espruino/BangleApps #3785: [launcher] clocks/fullscreen settings don't work, some font sizes break everything

onebitboy, 2025-03-28:
### Affected hardware version

Bangle 2

### Your firmware version

2v25 and 2v25.366

### The bug

1. Launcher seems to completely ignore the "Show clocks" and "Fullscreen" settings. No matter what it's set to, it always shows clocks and never goes fullscreen. I've tried it with v0.21 and v0.22 (stable and dev branches) on both the 2v25 and the 2v25.366 firmware.

2. Some launcher font sizes cause a complete breakage at least on 2v25.366. Setting it to (non-vector) 17 for example, the launcher doesn't display anything any more at all, so you can't even go back to the settings to change the font size again.

dashavoo, 2025-03-28:
Aside: This should probably be two separate issues.

> Launcher seems to completely ignore the "Show clocks" and "Fullscreen" settings. No matter what it's set to, it always shows clocks and never goes fullscreen. I've tried it with v0.21 and v0.22 (stable and dev branches) on both the 2v25 and the 2v25.366 firmware.

I didn't know about the fullscreen option, so when I saw this bug open I was interested to try it.
Like for @onebitboy, on my watch the setting seemed to make no difference. However, if I run the launcher from the IDE (Connect to watch, press the Storage icon , find `launch.app.js` and press the "Run File" icon) the launcher loads and correctly does not display widgets. It continues to work for as long as the watch doesn't reset (so I can move between launcher and fast-load supporting clock as many times as I like and the launcher doesn't show widgets, but if I load another app such as 
"Settings" the launcher no longer seems to respect the settings.

dashavoo, 2025-03-28:
@onebitboy which Clock are you using please?

dashavoo, 2025-03-28:
Aha.

The clock I use supports fast loading. I switched to one that doesn't and the fullscreen option worked correctly. So I checked the instructions for supporting fast loading.

I think [this](https://www.espruino.com/Bangle.js+Fast+Load#taking-advantage-of-fast-loading) is the paragraph that explains this behaviour:

> You can also use Bangle.showClock() (if your app is not a clock - eg a launcher). Or, you can use Bangle.load("load_me.app.js") - but again, **the app you load must use widgets, since your Clock would have loaded with widgets and they cannot be unloaded.**

onebitboy, 2025-03-29:
I noticed launcher just got updated to v0.23. The fullscreen bug is still present though (tested on firmware 2v25.369).

> Aside: This should probably be two separate issues.

I'll create separate issues for the "show clocks" setting and the font breakage and edit this one. I originally assumed it might all be related, but since this seems to be not the case, I agree that splitting it up is better.

thyttan, 2025-03-29:
I've added some code to fix both problems. Please try here: https://thyttan.github.io/BangleApps/?id=launch

onebitboy, 2025-03-29:
Works. Thank you! Maybe the option should be renamed to "Hide widgets", since the launcher still won't use the widget area of the screen even when they're hidden.

thyttan, 2025-03-30:
> since the launcher still won't use the widget area of the screen even when they're hidden.

But it does use the whole screen on my watch now. Doesn't it on yours?

onebitboy, 2025-03-30:
No, it just stays empty. I've attached a screenshot.

[image]

closed 2025-04-01 by gfwilliams