espruino/BangleApps #3786: [launcher] "show clocks" setting doesn't work

onebitboy, 2025-03-29:
### Affected hardware version

Bangle 2

### Your firmware version

2v25.369

### The bug

Launcher completely ignores the "show clocks" setting. The installed clocks are always listed, no matter whether the setting is enabled or not.

### Installed apps

Tested with launcher 0.21, 0.22 and 0.23 on firmware 2v25, 2v25.366 and 2v25.369.

onebitboy, 2025-03-29:
Setting works again with your PR. Thanks!

thyttan, 2025-03-30:
I moved the fix for this from the app code to the settings code. If you want to try it again. Instead of adding comlexity to the app it's simpler to just clear the cached app info if the setting is toggled.

onebitboy, 2025-03-30:
Tested again. Still works :)

PR #3788 (2025-04-01): launch: fix fullscreen and showClocks setting
0.24: Fix fullscreen when fastloading the launcher. (TODO:fix back btn flicker)
      Fix respecting the showClocks setting.

Try it via my app loader: https://thyttan.github.io/BangleApps/?id=launch

closes #3786 and #3785

@gfwilliams because it is a core app.
closed 2025-04-01 by gfwilliams