espruino/BangleApps #3787: [launcher] Some font sizes softbrick the launcher

onebitboy, 2025-03-29:
### Affected hardware version

Bangle 2

### Your firmware version

2v25.369

### The bug

Setting the font size to non-vector sizes bigger than the default 12x20 completely breaks launcher. It only shows a blank screen, so opening the settings again to revert the font size is impossible. launcher has to be uninstalled and then reinstalled again to fix it.

### Installed apps

Tested with launcher 0.21, 0.22 and 0.23 on firmware 2v25.366 and 2v25.369.

bobrippling, 2025-03-31:
@gfwilliams not sure if this is an old bug or possibly from some of the more recent font changes

gfwilliams, 2025-03-31:
Thanks - just putting a fix in now.

Looks like it wasn't my latest changes to the launcher that broke it, but adding extra fonts to the new builds broke it because it was expecting every font to have an `x` in the name!

commit 1c85059d (2025-03-31): 0.24: Fix Launcher when a custom font from 2v26+ is specified (fix #3787)