espruino/BangleApps #3792: [Bangle.JS App Loader] Backing up fails due to invalid JSON error

seancaulfield, 2025-04-01:
### Affected hardware version

Bangle 2

### Your firmware version

2v25

### The bug

When attempting to backup my Bangle.JS from the app loader, I get a JSON decode error. From the browser console:

```
<UART> Unable to decode "eval(process.env.CONSOLE).println(JSON.stringify(require(\"Storage\").list(undefined,{sf:false}).map(encodeURIComponent)))\r\n[\"about.img\",\"health.img\",\"setting.img\",\"sched.img\",\"alarm.img\",\"gpsinfo.img\",\"barometer.img\",\"barometer.info\",\"magnav.img\",\"magnav.json\",\"info.img\",\"info.info\",\"mylocation.json\",\"launch.json\",\"sched.settings.json\",\"antonclkplus.img\",\"random.img\",\"antonclk.json\",\"widhrm.info\",\"hralarm.json\",\"widhrm.json\",\"sched.json\",\"messagegui.img\",\"widlock.info\",\"wid_a_battery_widget.info\",\"widbt.info\",\"recorder.img\",\"hrm.img\",\"thermom.img\",\"thermom.info\",\"ios.img\",\"widalarm.info\",\"about.info\",\"magnav.info\",\"chimer.info\",\"antonclkplus.info\",\"random.info\",\"mylocation.info\",\"ios.info\",\"hrm.info\",\"gpsinfo.info\",\"sched.boot.js\",\"messagegui.new.js\",\"messagegui.boot.js\",\"recorder.clkinfo.js\",\"recorder.settings.js\",\"notify.info\",\".bootcde\",\"widChargingStatus.info\",\"messages.settings.json\",\"ios.settings.json\",\"recorder.json\",\"messageicons.info\",\"reply.info\",\"recorder.info\",\"barometer.app.js\",\"info.app.js\",\"widpedom.wid.js\",\"widpedom.settings.js\",\"widpedom.info\",\"widhrm.wid.js\",\"widhrm.settings.js\",\"widlock.wid.js\",\"wid_a_battery_widget.wid.js\",\"widbt.wid.js\",\"thermom.app.js\",\"widalarm.wid.js\",\"about.app.js\",\"magnav.app.js\",\"magnav\",\"chimer.wid.js\",\"chimer.settings.js\",\"antonclkplus.app.js\",\"antonclkplus.settings.js\",\"random.app.js\",\"mylocation.settings.js\",\"ios.app.js\",\"ios.boot.js\",\"ios.settings.js\",\"hrm.app.js\",\"health.settings.js\",\"gpsinfo.app.js\",\"notify\",\"widChargingStatus.wid.js\",\"messageicons\",\"messageicons.img\",\"reply\",\"recorder.app.js\",\"recorder.wid.js\",\"widmessages.wid.js\",\"widmessages.info\",\"alarm.app.js\",\"alarm.settings.js\",\"alarm.info\",\"bootupdate.js\",\"boot.info\",\"sched.js\",\"sched\",\"sched.settings.js\",\"sched.clkinfo.js\",\"sched.info\",\"messages\",\"messages.settings.js\",\"messages.info\",\"messagegui\",\"messagegui.app.js\",\"messagegui.info\",\"font\",\"fontext.boot.js\",\"fontext.pbf\",\"fontext.info\",\"chimer.json\",\"health.json\",\"health-2025-4.raw\",\"setting.json\",\"launch.app.js\",\"launch.settings.js\",\"launch.info\",\"health.app.js\",\"health.boot.js\",\"health\",\"health.info\",\"setting.app.js\",\"setting.info\",\"wpedom.json\",\"launch.cache.json\",\".boot0\",\".widcache\",\".clkinfocache\"]\r\n=undefined\r\n>", got SyntaxError: Unexpected token 'e', "eval(proce"... is not valid JSON
```

No idea what to do to further debug here.

### Installed apps

barometer (0.04), info (0.03), widhrm (0.06), widlock (0.08), wid_a_battery_widget (0.04), widbt (0.09), thermom (0.07), widalarm (0.02), about (0.15), magnav (0.07), chimer (0.05), antonclkplus (0.11), random (0.03), mylocation (0.11), ios (0.18), hrm (0.13), gpsinfo (0.13), notify (0.14), widChargingStatus (0.04), messageicons (0.08), reply (0.03), recorder (0.44), widpedom (0.22), widmessages (0.07), alarm (0.50), boot (0.65), sched (0.30), messages (0.65), messagegui (0.84), fontext (0.03), launch (0.22), health (0.31), setting (0.76)

bobrippling, 2025-04-02:
Seems like the watch is echoing back what it's sent - @gfwilliams have you seen this before?

gfwilliams, 2025-04-02:
Ahh, yes - it was posted in the forum a few days ago: https://github.com/orgs/espruino/discussions/7717

I believe it's now fixed in the development app loader (I thought that was as far as the problem had got), but I've now just updated the main banglejs.com/apps so it should be fixed there now too.

gfwilliams, 2025-04-15:
Ok, thanks - that's really helpful. I've been struggling to reproduce it but trying to upload `sched` lets me get it to happen. I added "File Upload Compatibility mode (disables binary packet upload)" under `More...` earlier - can you confirm that fixes it if it's enabled?

bobrippling, 2025-04-15:
Ah great - yes - "File Upload Compatibility mode" fixed it 🎯

gfwilliams, 2025-04-15:
Argh - it's been pretty hard to reproduce!

I *think* it happens when we're in a low speed BLE connection. Espruino has a 1 second timeout on packet transfers, and it seems like if BLE is in low speed mode (you can force it with `NRF.setConnectionInterval(200)`) or BLE compatibility mode then a 1kb packet can take longer than that to transfer, and so Espruino NAKs.

Does that sound likely to you? Maybe you have the compatibility mode set? Or I guess if you stay connected by BLE and then don't do anything for a minute Espruino goes into low speed/power mode and maybe that could kick it off (but next time you tried the upload it should be in high speed mode and would work)

closed 2025-04-15 by gfwilliams

bobrippling, 2025-04-15:
> Does that sound likely to you? Maybe you have the compatibility mode set? Or I guess if you stay connected by BLE and then don't do anything for a minute Espruino goes into low speed/power mode and maybe that could kick it off

Hard to say - I don't have compatability mode / 20-byte limit enabled, as far as I'm aware it's using "factory" bluetooth settings. I'd do the app update almost immediately after connecting so hard to say what's causing it. There's a lot of bluetooth noise where it was, so could be that?

gfwilliams, 2025-04-16:
>  There's a lot of bluetooth noise where it was, so could be that?

I guess so - it could be a combination of things. I had figured that normally it negotiates ~120b MTU at 130hz so gets 15k/sec, so trying to get 2k down (it was 2k, not 1k) wasn't unreasonable. But if it doesn't negotiate a higher MTU 130*20=2600b which I guess is getting close enough that it can fail especially if as you say there's a bunch of noise.

bobrippling, 2025-04-16:
I'm able to complete the `sched` update with no problem in a less noisy environment, so perhaps it is that.

However my nightly backup of my watch failed this morning - woke up to my watch frozen and needed a hard reset to bring it back. Could that be related? Might my backup (which is entirely old-school and ascii) have triggered something?

gfwilliams, 2025-04-16:
> However my nightly backup of my watch failed this morning - woke up to my watch frozen

Hmm, I'm not sure - I was pretty sure all the backup stuff (if using the app loader?) still just uses the old-style transfers which haven't changed. Maybe next time it happens see if you can connect with the Web IDE - it might just be that the backup uses Ctrl-C and that broke out of the interval that updated the clock?

bobrippling, 2025-04-16:
Ah no - this was my own backup code talking over the UART characteristic. I'll see if I can get more details next time it happens