espruino/BangleApps #3804: [alarm/sched] Timer doesn't fire (or fires immediately) when enabling from list

Blockguy24, 2025-04-17:
### Affected hardware version

Bangle 2

### Your firmware version

2v25

### The bug

### Description
On the Bangle.js 2, a timer can be quickly enabled by pressing its icon or long pressing the item in the list. However, this causes some unexpected behaviour. Most of the time the timer never fires, but sometimes it fires immediately after enabling it this way.

### Expected Behaviour
Enabling the timer via this shortcut should have the same effect as opening its menu and setting it to be Enabled.

### Suspected cause
When enabling via the main menu, neither `prepareTimerForSave` (or `prepareAlarmForSave` for alarms) is called, so an old expiration time is used for the timer instead of it being updated when it is enabled.

### Installed apps

- alarm (0.50)
- sched (0.31)
- boot (0.65)

bobrippling, 2025-04-17:
Seems to me like the fix would be to replicate this code (in `alarm`):

https://github.com/espruino/BangleApps/blob/cc051726ccd43ca079cf7e006d9571d87e5676f3/apps/alarm/app.js#L543-L544

into the clkinfo, or more specifically, into `sched`'s `resetTimer()`:

https://github.com/espruino/BangleApps/blob/cc051726ccd43ca079cf7e006d9571d87e5676f3/apps/sched/lib.js#L44-L49

Essentially setting `alarm.last = 0` (with a few tweaks).

Does #3805 resolve it for you? I can deploy the code if that'd help, I just have another deployment in use for #3803 atm.

Blockguy24, 2025-04-18:
I tried the update but it didn't really change much, the timer will still either not fire or fire immediately

bobrippling, 2025-04-18:
Thanks - that mustn't be the bug, although I think you're right that it's related to `.last`

bobrippling, 2025-04-24:
@Blockguy24 when you say
> a timer can be quickly enabled by pressing its icon or long pressing the item in the list

Which list do you mean? I took it to be the clock info, but do you mean the alarm menu?

Blockguy24, 2025-04-24:
Yeah, the first thing that appears when opening the Alarms app. I see a list of all alarms, timers and events, and I can long press or tap the icon of the alarm/timer to toggle it enabled or disabled. However, enabling timers in this way has the weird behavior I described.

bobrippling, 2025-04-25:
Ok - tracked that down, could you let me know if #3805 now fixes it for you?

Blockguy24, 2025-04-26:
No, the issue seems to still persist somehow.

bobrippling, 2025-04-28:
Trying it out, I don't see the long-press (or icon) touch behaviour at all, in fact. I just get booted into the edit menu. @gfwilliams is this still intended behaviour?

thyttan, 2025-04-28:
Here's related stuff: 
- https://github.com/espruino/BangleApps/pull/3604
- https://github.com/espruino/Espruino/pull/2565
- https://github.com/espruino/BangleApps/pull/3609
- https://github.com/espruino/Espruino/pull/2569

gfwilliams, 2025-04-28:
> @gfwilliams is this still intended behaviour?

I'm not sure - I think best to leave this to  @thyttan 

But I just tested on a watch here.

* Tapped `New` then `Timer`, `Confirm` (makes a 5m timer)
* Long-pressing on the `5m` appears to toggle the timer state after a second or so

`require("sched").getAlarms()` reports:

```
>s.getAlarms()
=[
..
  { timer: 300000, del: true, on: true, rp: false,
    as: false, dow: 127, last: 0,
    vibrate: "::",
    t: 41318000 }
 ]
>s.getAlarms()
=[
..
  { timer: 300000, del: true, on: false, rp: false,
    as: false, dow: 127, last: 0,
    vibrate: "::",
    t: 41318000 }
 ]
```

no matter how many times I do it, so `t` isn't getting updated - probably what @Blockguy24 said originally about `prepareTimerForSave` being missing? Looks like we normally do it when enabling timers

gfwilliams, 2025-04-28:
Do you want me to do this or were you looking into it @thyttan? Looks to be a reasonably easy addition

closed 2025-04-28 by gfwilliams

gfwilliams, 2025-04-28:
Cool - just fixed I think.

Sorry @bobrippling I just realised it's basically what you'd modified #3805 to do - for some reason I saw that and thought it had been merged already. It definitely seems to fix things for me.

@Blockguy24 maybe try installing from the development app loader now. Toggling definitely seems to change `t` in `require("sched").getAlarms()` which is what you'd hope, and the timer fires when I expect it.

Blockguy24, 2025-04-28:
Ah yes, this does appear to be fixed now. Thank you everyone!

I did try to deploy the fix earlier myself from the PR, must've done something wrong because it works fine when I updated from espruino.github.io

commit 0a676195 (2025-04-28): 0.51: Fix long-touch to enable alarm/timer not updating time (fix #3804)
PR #3805 (2025-04-29): sched: prevent timers from firing immediately
fixes #3804