espruino/BangleApps #3818: [warpdrive] App Soft-Bricked my Watch

voloved, 2025-04-21:
### Affected hardware version

Bangle 2

### Your firmware version

2v25...I think? My watch is a brick so I can't check

### The bug

I installed the warpdrive clock face.
I set it as my default clock. I move to the clock face, and it shows `[LOADING]`.
I installed it at 7:48am. It is currently 7:58am and for the past 10 minutes, the screen has not changed.
I held the physical button and nothing changed. The backlight has been on the whole time and the time at the bottom of the face shows 7:48. I tried connecting to bluetooth, but it's not detecting on my computer anymore. 

[Where Loading shows up in the warpdrive app](https://github.com/espruino/BangleApps/blob/master/apps/warpdrive/app.js#L591)

voloved, 2025-04-21:
After multiple attempts of resetting by holding the button, it finally reset about 20 minutes after getting into that mode.
I uploaded the warpdrive face to RAM via the IDE to track errors on the physical watch. Error message and image of the IDE is below.
After that message, the watch disconnected from BLE and is unresponsive again.

`Found lcdBuffer at 20031cfc stride=68`

[image]

thyttan, 2025-04-21:
Very weird that the watchdog on the hardware button doesn't work reliably!

I think this may be one for @gfwilliams when he gets back.

gfwilliams, 2025-04-22:
Thanks for the report - this is very strange. Does anyone else see problems? I just tried it here and it appears to work fine - `[LOADING]` appears for maybe half a second at most.

You mention installing Power Manager - I guess you have other apps/widgets installed as well? I guess one option is that you had enough extra things running that there wasn't enough memory for a flat data buffer to be allocated. I notice that this app doesn't seem to check if `E.getAddressOf` returns a valid number (or 0).

Please could you add `print(process.memory())` where you had `print("b")` - and maybe also print the value returned by `E.getAddressOf(sintable, true)` and `E.getAddressOf(translation, true)`

How long did you hold the button down for? Bangle.js has a hardware watchdog which is almost impossible to disable if code isn't running - if everything locks up or you hold the button down for 5-10 seconds then it doesn't just reload the app, some hardware in the watch triggers a hardware reset. I'd be amazed if that wasn't working (unless the watch wasn't entirely locked up and you'd installed some other app/code that disables it)

voloved, 2025-04-24:
I added the printing of process.memory, but failed to print the address locations.
I have quite a few programs running on it, so the limit of memory can be an issue.

Right now, the watch is stuck again. I hold the button for >15 seconds. I've tried doing this 5 times in a row.
Bluetooth to IDE also disconnected.

I think I will stop testing this on hardware; the last time I had this issue, I needed to deplete the battery to 0% so it'll clear the program from RAM.

[image]

voloved, 2025-04-25:
Long pressing resets without issue.
Below is running the code you shared, where I pressed the button twice.
I'll send the backup.
I did notince a large amount of trash in my memory. Is that normal? If not, how do I clean it?

```_                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v25 (c) 2024 G.Williams
>
>E.kickWatchdog()
Uncaught Error: Watchdog kicked
 at line 1 col 26
throw new Error("Watchdog kicked");
                                 ^
in function "kickWatchdog" called from line 1 col 16
E.kickWatchdog()
               ^
>E.kickWatchdog
=function () { ... }
>Button held down - interrupting JS execution...
Execution Interrupted
>
```

[image]

gfwilliams, 2025-04-25:
Fixed it! I got the backup and have managed to reproduce. Cutting edge firmwares (so 2v26 when released) do auto-reboot rather than hang so that's something.

... but the issue was you had the watch in left-handed mode, and the app was getting confused while trying to find the buffer to write to and so was writing from the end of the screen buffer onwards! 

If you upload to warpdrive 0.04 from the development app loader then it should all be sorted now though

commit 4b9fae38 (2025-04-25): warpdrive 0.04: Remove buffer finding code (which fails if screen rotated) and make dependent on fw 2v21's Bangle.getOptions().lcdBufferPtr (fix #3818)
Also adds check for non-contiguous memory
commit e867ff5c (2025-04-25): synthwave 0.04: Remove buffer finding code (which fails if screen rotated) and make dependent on fw 2v21's Bangle.getOptions().lcdBufferPtr (fix #3818)
closed 2025-04-25 by gfwilliams