espruino/BangleApps #3824: [Settings] Error compacting storage: Can't read property 'scrolll' of undefined

banglejsdev, 2025-04-24:
### Affected hardware version

Bangle 1

### Your firmware version

2v25

### The bug

In the Settings app on Bangle 1 (but not Bangle 2), using Utils | Compact Storage always results in the following error appearing in the IDE:
```
>Compacting...\
Uncaught Error: Can't read property 'scroll' of undefined
 at line 3 col 2909 in setting.app.js
...enu[""].scroll=menuScroller.scroll;menuScroller=E.showMenu(m...
                              ^
in function "restoreMenu" called from line 18 col 69 in setting.app.js
...ct();restoreMenu(utilMenu());
                              ^
in function "a" called from line 1 col 187
...ected))}else if("function"==typeof a)a();else if("object"==typeof a){if("number"==typeof a.v...
                                                              ^
in function "select" called from line 1 col 22
a?d.move(a):d.select()
                     ^
in function "d" called from line 1 col 3
d()
  ^
in function called from system
>

```
The Bangle 1 appears to hang because the display with the "Compacting..." message remains, but in fact after compacting storage finishes, pressing BTN1 or BTN3 overwrites the display with selections of the Utils menu.

The same behavior/error also occurs when doing Utils | Turn Off | No

thyttan, 2025-04-24:
I believe @bobrippling will be best equipped to fix this? :)

bobrippling, 2025-04-25:
Yeah, think it's a simple fix. @banglejsdev are you able to test the change with your bangle 1? I have the new settings app [deployed here](https://bobrippling.github.io/BangleApps/?id=setting)

banglejsdev, 2025-04-25:
Thanks, I tested your version of the app and it appears to fix the problem.

However, your version of Settings might have other issues, perhaps due to my missing other apps that also need to be updated on my Bangle 1. In particular, if I press the red "back arrow" from within the launcher now, I get the following error:

```
 2v25 (c) 2024 G.Williams
Uncaught Error: Function "setLCDOverlay" not found!
 at line 1 col 458 in launch.app.js
exports.offset=-24,Bangle.setLCDOverlay(undefined,{id:"widget_utils"...
                                  ^
in function "cleanupOverlay" called from line 1 col 731 in launch.app.js
exports.cleanupOverlay(),delete exports.offset,exports.swipeHandle...
                             ^
in function "cleanup" called from line 1 col 291 in launch.app.js
if(exports.cleanup(),!global.WIDGETS)return;for(var a of global.WIDGETS){if...
                                 ^
in function "show" called from line 2 col 219 in launch.app.js
...quire("widget_utils").show();
                              ^
in function "b" called from line 1 col 502
...iRemove;delete Bangle.uiRemove;b()}g.reset();if(c){if("updown"==c)Ban...
                                       ^
in function called from line 1 col 31
Bangle.uiRemove?(Bangle.setUI(),(__FILE__=a)||(a=".bootcde"),...
                               ^
at line 1 col 130
...a))):load(".bootcde"!=a?a:void 0)
                                   ^
in function called from line 1 col 23
Bangle.load(".bootcde")
                      ^
at line 1 col 23
Bangle.load(".bootcde")
                      ^
in function called from line 1 col 12
1==b&&a.back()
             ^
at line 1 col 12
1==b&&a.back()
             ^
in function called from system
```

bobrippling, 2025-04-25:
Interesting - that looks unrelated. Do you see it without my fix? (let me know if you need instructions on backing out the fix)

banglejsdev, 2025-04-25:
Oh, sorry, you're right--since the error occurs in the launcher code it appears unrelated to your fix. I usually use the red "back arrow" to return to the clock from the launcher and have never seen this issue before so believe it's something that started happening after I tested your change. If I remember correctly, when I installed your updated Settings app there were a couple of other apps that also showed up as being outdated, so I think I may have updated those as well. Unfortunately, I don't remember which ones those were--the Launcher and Bootloader, perhaps?
In any case, I went back to the official App Loader, uninstalled Settings, Launcher, and Bootloader and reinstalled the current versions (v0.25, v0.65, and v0.76, respectively) and am still seeing this red "back arrow" problem. I see that the launcher is now showing more entries per page than it used to some other launcher settings must have gotten updated as well...
If there's something else I should do to back out your change please let me know. Otherwise, despite my Launcher acting a little strangely now I can still use the apps I use my Bangle 1 for, so no worries.

Thanks for fixing the original problem I wrote up!

thyttan, 2025-04-25:
I found out at https://github.com/espruino/BangleApps/pull/3685#issuecomment-2679938571 that "Widget Utils seems to have become incompatible with Bangle.js 1 at some point because of it's reliance on Bangle.setLCDOverlay."

It's seems I didn't go to fix it in widget_utils at that time. I opened an issue to track that now at #3830.

PR #3828 (2025-04-26): settings: handle a `undefined` scroller (BJS1)
Fixes #3824
commit 80cc3a60 (2025-04-26): settings: handle a `undefined` scroller (BJS1)

Fixes #3824
closed 2025-04-26 by thyttan