espruino/BangleApps #3842: Settings v0.78 fails to start after installation

nxdefiant, 2025-05-04:
On Firmware 2.v25 I get the following error when starting settings:

Error:
> Uncaught SyntaxError: Got EOF expected '}'
 at line 21 col 1890 in setting.app.js
...;}});}pushMenu(mainMenu());

thyttan, 2025-05-04:
> only the minified version installed from the development apploader fails.

Do you mean just this setting on:

[image]

Or this one:

[image]

?

I also get the same error as you with the first setting above on, but the second off.

@gfwilliams I wonder if it can be something with https://github.com/espruino/BangleApps/commit/04023e859bfcc00a06dd0fd6a0b05f68e7b28535 and module pretokenisation?

nxdefiant, 2025-05-05:
Sorry with minified I meant the stripped version that is uploaded to the watch, yes, I have pretokenise=true, minify=false (Which is default I guess?)

bobrippling, 2025-05-05:
Yeah I'd suspect if it works ok untokenised then the bug is likely there

gfwilliams, 2025-05-06:
Hmm, ok, thanks. I'll look into this today. So this is on the development app loader?

I was pretty sure this worked on a cutting edge build, so maybe the issue is that the pretokeniser is trying to pretokenise ints (which only the latest Espruino builds support)? It's supposed to check the version but maybe that isn't working

gfwilliams, 2025-05-06:
Please can you try now? It turns out it was the minifier that had been changed based on some linting tweaks, and it had broken the minification of the date_utils module (but not other most modules it seems)

thyttan, 2025-05-06:
Fixed for me! Reinstalled via development app loader in Gadgetbridge. No error messages in app loader terminal.

gfwilliams, 2025-05-06:
Great - Fixed with https://github.com/espruino/EspruinoTools/commit/2e26fdeabb0b806e101d975f951342ce307497d8 then