espruino/BangleApps #3848: [android/http] Add flag to http intent allow self signed certificates

samakj, 2025-05-12:
### Affected hardware version

Bangle 2

### Your firmware version

2V25

### The bug

Currently making fetch requests to a https server using unsigned certificates (in my case a locally hosted instance of home assistant behind traefik proxy) results in:

`CertPathValidatorException: Trust anchor for certification path not found`

(I would imagine) Self signed certificates are fairly common in the homelab/self hosting community and so it would be amazing to make http requests to servers using them.

My suggestion would be to add `insecure: true` to the http intent and update gadgetbridge http handling accordingly.

Gadgetbridge version: 0.85.0-banglejs

bobrippling, 2025-05-14:
@gfwilliams another one for you

gfwilliams, 2025-05-14:
I think this would be a great idea. In a way I feel like we could probably almost have it allowing insecure by default (I'd have loved to have it just work on HTTP too, but I believe I'd have to get extra permissions in the Play Store for that).

@samakj you sound like you know what'd be needed on the Android side to make this work? If so it'd be great if you could add it to Gadgetbridge and submit them a PR. The relevant code for the HTTP request is: https://codeberg.org/Freeyourgadget/Gadgetbridge/src/commit/488c789e152cbc3fefbda7444d9cfceab274071e/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/banglejs/BangleJSDeviceSupport.java#L940

samakj, 2025-05-14:
I think insecure by default sounds like considerable security hole to me but for those that know what they are doing and likely host many of their own services behind self signed certs or just want to spin up a server quickly for local testing it would be an amazing upgrade.

@gfwilliams unfortunately I am not familiar with java, i know python requests has `verify=False` and node has something similar, so I assume java has this functionallity to (?).

https://stackoverflow.com/questions/875467/java-client-certificates-over-https-ssl/876785#876785 this looks promising

P.S. Loving the project and working on my own clock, may look into some home assistant stuff if we can get this feature. I use tailscale on my phone to connect to home assistant remotely so it would be pretty cool to be able to monitor things remotely via my watch using http intents

gfwilliams, 2025-05-14:
Adding an option for it probably isn't too bad - you can see from how other options are in that code how it'd be done. But given this is something that is disabled by default anyway and is used pretty rarely, allowing insecure HTTPS is probably not a huge deal as long as it's documented.

> unfortunately I am not familiar with java,

If I'm honest, I'm not that familiar with Java (at least under Android) either - and I don't have all that much time at the moment.

So if you really want this, maybe give it a stab? The Gadgetbridge folks are really helpful at suggesting when they think something is a bit off with a PR.

samakj, 2025-05-14:
Ok, if i find some time i will see if I can make a pull request. If you do find some time to do it before me then please let me know 😄

samakj, 2025-05-18:
The kind people at gadget bridge have sorted this:

https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/4845

https://github.com/espruino/EspruinoDocs/pull/749

Ill close this issue when they merge it in 👍