espruino/BangleApps #3850: Apploader only connects once

nxdefiant, 2025-05-13:
With firmware 2v25 and chromium 134 when
1. clicking connect
2. clicking disconnect
3. clicking connect again the connection fails with

> Device connection failed, No response from device. Is 'Programmable' set to 'Off'?

Browser console:
>  comms.js:370 <COMMS> Ctrl-C gave ""
context-menu-content-scripts.js:25 Using workaround to get event.path
comms.js:431 No JSON, just got: "\n"
ui.js:76 <TOAST>[error] Device connection failed, No response from device. Is 'Programmable' set to 'Off'?
uart.js:1274 <UART> Disconnected (gattserverdisconnected)
uart.js:1274 <UART> Disconnected
uart.js:1274 <UART> Disconnected (gattserverdisconnected)
uart.js:1274 <UART> Disconnected
uart.js:1274 <UART> Disconnected (gattserverdisconnected)
uart.js:1274 <UART> Disconnected
uart.js:1274 <UART> Disconnected

bobrippling, 2025-05-13:
Tagging @gfwilliams

gfwilliams, 2025-05-14:
Thanks, I was looking at this yesterday. What OS are you on?

On Android it seems this works fine (not using Gadgetbridge), but I did hit this on Linux.

It's a bit of an odd issue because it seems that Web Bluetooth itself fails. We register a listener for when the bluetooth characteristic changes (eg data is received) but it just never gets called the second time it's added.

nxdefiant, 2025-05-14:
Linux it is.

gfwilliams, 2025-05-14:
Thanks for the quick response! So just some background... It's this line:

https://github.com/espruino/EspruinoWebTools/blob/31a824791576deed654a78bd2c11f907a310b6ef/uart.js#L919

It used to be that the 'old' listener got called, and that then put the received data into the wrong `connection` object. It only worked originally because multiple connections weren't handled properly.

But now I try and remove the old listener and add a new one, and the second one is never called. So I guess I'll have to have some kind of hack so we only call addEventListener one time even through multiple connections

closed 2025-05-14 by gfwilliams

nxdefiant, 2025-05-14:
Wonderful, as always

commit d6e465f4 (2025-05-14): uart 1.17: Work around Linux Web Bluetooth Bug (connect-disconnect-connect to same device)
fix #3850