espruino/BangleApps #3852: [andoid/music] Allow banglejs to request musicinfo/musicstate

samakj, 2025-05-14:
### Affected hardware version

Bangle 2, Bangle 1

### Your firmware version

2V25

### The bug

As it currently stands, to receive information about currently playing media on the connected gadget bridge mobile you have to wait for a `musicstate` or `musicinfo` event. This leads to some difficulty in displaying any sort of music controls as these events dont always occur together. e.g:

- user presses prev/next on the mobile phone, the bangle receives the `musicinfo` but no state so it is unknown if the music is playing or paused
- user presses play/pause on the mobile phone, the bangle receives the  `musicsctate` but no info so its not possible to display what is currently playing

The above makes it quite hard to develop a UI/UX that feels intuitive/comfortable because there is often some information missing/out of date

My suggested solution is to allow the bangle to send `musicinfo` and `musicstate` intents which would trigger the mobile phone to send back the equivalent event. 

This may be a gadgetbridge issue,  if so please let me know and I will open an issue over there instead.

thyttan, 2025-05-14:
Tagging @gfwilliams

gfwilliams, 2025-05-14:
I'm pretty sure that musicinfo/musicstate gets sent to the `messages` library which merges them and stores the current state (in a message with `id:"music"`) so if your app wants to get the current state it should be able to query that?

Obviously if the Bangle only just connected to Gadgetbridge they'd be out of date though - we should probably make Gadgetbridge send the current music state when the Bangle is connected.

Adding a feature to query music state to Gadgetbridge (if the data is available there to query) could be a neat idea, but I'm not entirely sure it's needed at the moment if the music states are being stored?

samakj, 2025-05-14:
I did not realise they are being stored, I suppose that goes some way to fixing the issue, relies on the messages app being installed though, would be nice to have it not rely on another app. 

Sending them on connection also sounds like a good idea 👍

gfwilliams, 2025-05-14:
> relies on the messages app being installed though

To be fair, if you have Gadgetbridge installed you're almost certainly doing it in order to also get messages? Installing the Android Integration app also installs messages automatically anyway (which IIRC is not the Messages app, just the library that handles loading+saving them)

samakj, 2025-05-14:
Ok fair enough, will close the issue 👍

closed 2025-05-14 by samakj