espruino/BangleApps #3891: [App Loader] Deletes app before confirmation is selected

RKBoss6, 2025-06-18:
### Affected hardware version

Bangle 2, Bangle 1

### Your firmware version

2v27

### The bug

In the App Loader, when deleting an app, a confirmation window pops up, but also deletes the app, before user inputs confirmation. Can be harmful if the delete button for an app is accidentally pressed, as confirmation does not work.

RKBoss6, 2025-06-18:
If someone tells me what file the delete app function is in, I should be able to fix this.

gfwilliams, 2025-06-19:
Thanks! Well spotted! Just merged and made this live - I made a minor tweak to bracketing because ideally we want the entire operation to be inside `startOperation`, rather than after it.

Sorry about that - that's a regression from the recent code that tries to stop operations from overlapping.

closed 2025-06-19 by gfwilliams