espruino/BangleApps #3915: Feature discussion: [Alarm/sched] Improve snoozing

thyttan, 2025-06-29:
I've been wanting a more fleshed out snoozing experience. 

AFAIK today we can adjust the length of snoozing in the sched settings and this applies equally to all alarms/timers/etc. In some situations this feels a bit clunky to me. A short timer for when I'm cooking may not beg for the same length snooze as is appropriate when waking up by an alarm. 

Some ideas: 
- when pressing "Snooze" open a new prompt with up to four options for length.
- treat snoozing for timers differently to alarms and events
  - add a separate snooze for timers setting
    - in addition to fixed length snooze add possibility to have snooze length be set as some fraction of the timer length in question.

Please suggest other ideas as well :)

thyttan, 2025-07-03:
@RKBoss6 I think that would be a simple and elegant solution! I don't know if `showPrompt` propagates long press information. If not we could possibly add that.

bobrippling, 2025-07-03:
I think we make it configurable and default to those ⬆ 

For long-press, we can probably `showPrompt` but have a `setWatch` listening for a long press and that listener will cancel the prompt (via a `Bangle.setUI()`)

RKBoss6, 2025-07-03:
Yes, I was imagining the hold for the software button. How would we go about doing that? I am somewhat new to the bangle.js workspace, so if anyone knows goes to achieve this, drop a reply.

bobrippling, 2025-07-04:
We can't listen for a long press on [`showPrompt` - the functionality isn't there](https://github.com/espruino/Espruino/blob/5fb49303f67cf85648324ad44a3f9eb53e7fcf89/libs/js/banglejs/E_showPrompt_Q3.js), but using [a listener on the screen](https://github.com/espruino/BangleApps/issues/3915#issuecomment-3032464458) in general would work - we'd want a [`"touch"` handler](https://www.espruino.com/ReferenceBANGLEJS2#l_Bangle_touch):

```javascript
Bangle.prependListener("touch", (_btn, xy) => {
  if (xy.type === 2) {
    // long press - show the snooze menu
  }
})
```

thyttan, 2025-07-04:
Yes and we could even limit it to the area of the snooze button 👍I think that would be neat.

RKBoss6, 2025-08-22:
Although, when we do merge this, users will need the latest cutting-edge firmware for it to work. Maybe we can just add a check for compatibility, and if not present, just keep default snooze behavior... If the fw supports the long press, then use the new snooze menu.

thyttan, 2025-08-22:
The way I intend to tweak `showPrompt` that should be taken care of there automatically 😇

bobrippling, 2025-08-22:
Yes, it's a bit difficult while keeping the existing interface. Currently we have:

```typescript
showPrompt<T>(
  message: string,
  options: {
    title?: string,                      
    buttons?: {
      [name: string]: T, // the `T` is returned
    },
    ...
  },
): Promise<T>;
```

Perhaps for a long press, we allow a user to opt in like so:
```typescript
showPrompt(
  longPress: true,
  message: string,
  options: {
    title?: string,                      
    buttons?: {
      [name: string]: T, // the `T` is returned inside an object
    },
    ...
  },
): Promise<{ value: T, longpress: boolean }>;
```

The key difference being if `longPress: true` is passed as an option, the user gets back an object, `{ value: T, longpress: boolean }`, which still gives them their value, but also whether it was long pressed or not.

<details><summary> The final typescript looking like this: </summary>

```typescript
showPrompt(); // clear the UI

showPrompt(
  longPress?: false,
  message: string,
  options: {
    title?: string,                      
    buttons?: {
      [name: string]: T, // the `T` is returned inside an object
    },
    img?: string,
    remove?: () => void,
    buttonHeight?: number,
  },
): Promise<{ value: T, longpress: boolean }>;

showPrompt(
  longPress: true,
  message: string,
  options: {
    title?: string,                      
    buttons?: {
      [name: string]: T, // the `T` is returned inside an object
    },
    img?: string,
    remove?: () => void,
    buttonHeight?: number,
  },
): Promise<{ value: T, longpress: boolean }>;
```

</details>

bobrippling, 2025-08-26:
I think I follow, so we'll have something like:

```javascript
E.showPrompt({
  buttons: { Ok: 1, No: 2 },
  buttonsLong: { No: 2 },
})
  .then(answer => {
    // answer is 1 or 2
  })
```

So we still have the promise resolving with `1` or `2`:

```diff
           draw(i); // highlighted button
           g.flip(); // write to screen
           E.showPrompt(); // remove
+          if (e.type===2 /*long press*/ && options.buttonsLong && options.buttonsLong[btns[i]]) {
+            resolve(options.buttonsLong[btns[i]]); // options.buttonsLong[...] is 1 or 2
+          }
           resolve(options.buttons[btns[i]]);
         }
       });
```

So we still have the problem of distinguishing a long press, or am I missing something?

thyttan, 2025-08-26:
```
E.showPrompt({ buttons: { Ok: 1, No: 2 }, buttonsLong: { No: 3 }, }) .then(answer => { // answer is 1 or 2 or 3})
```

👆That way you can do something specific with case 3 (long press on "No"-button). 👌🤓

thyttan, 2025-08-26:
I realize I probably misunderstood and didn't really answer your question:

> but if there's a long press how does the user tell?

I guess in my suggestion there's no tell in UI for when a button has special long press action. I think we could do with adding notes to readme? It may be fun for a user to stumble upon it also? Like an easter egg :p

But we could experiment with an extra underline or dot as idicator or something? I think another shade of color is not easy w/o dithering which I think should be avoided.

RKBoss6, 2025-08-26:
I mean, the long press action should tell if it is a long press... eg: opening the snooze menu... we could do a color change when a long press is detected, but I really don't feel it's necessary. The user should now by how long they hold the button, and seeing the result from it.

gfwilliams, 2025-08-27:
This seems like a good plan for me. I'm happy to put the showPrompt change in.

Docs for this in the `sched` readme at least would be vital though. There's no point having functionality that only the authors of the PR know about :)

PR #3978 (2025-08-30): [Sched] Add snooze menu
This resolves #3915, adding a snooze menu if you long press the snooze button. This will only work if you upload this code to the bangle in the IDE, as a boot.js file, or install the latest cutting edge firmware: 
```
E.showPrompt=(function(message,options) {
  if (!options) options={};
  if (!options.buttons)
    options.buttons = {"Yes":true,"No":false};
  var btns = Object.keys(options.buttons);
  if (btns.length>6) throw new Error(">6 buttons");
  var btnPos;
  function draw(highlightedButton) {
    g.reset().setFontAlign(0,0);
    var R = Bangle.appRect, Y = R.y, W [... cut]
closed 2025-08-30 by gfwilliams