espruino/BangleApps #3928: [Messages] Smaller font for displaying messages

Poolitzer, 2025-07-07:
I  would like to suggest making the font smaller. I have good eyes, I can read that, so why not.

<p align="center">
  [image]
  [image]
</p>

(changed bodyFont to fontMedium [here](https://github.com/espruino/BangleApps/blob/6f309a88ae26a836457c8fb2a44c6bce5999079e/apps/messagegui/app.js#L365)).

And tbh I dont even need the large name, lets make it medium too.

<p align="center">
  [image]
  [image]
</p>

(changed titleFont to fontMedium [one line higher](https://github.com/espruino/BangleApps/blob/6f309a88ae26a836457c8fb2a44c6bce5999079e/apps/messagegui/app.js#L364)). Now the blue title box could even be decreased, though I am already procrastinating too much and wont try to hack this further. 

I can however understand lots of people like larger font, thats fine. I do personally really like this way of having more information viewable at once, so maybe this could be a setting option?

thyttan, 2025-07-07:
The body font size is dynamic though so a long message already gets the smaller font. Is that not the case for you?

thyttan, 2025-07-07:
Ah - you want the body font even smaller than in the screenshots?

I think that is the smallest today, as you indicate. But shorter messages will get a bigger font. I thought you had changes the font for short messages 😇

Poolitzer, 2025-07-08:
I mean thats fontMedium, in theory it could go to fontSmall. But thats really small.

I can not remember the font going from large to medium. Will test

Poolitzer, 2025-07-08:
Okay it works for me, no idea why I thought it didn't previously. Sorry for wasting time.