espruino/BangleApps #3998: [System] Loading boot0 on mundane operations

voloved, 2025-09-13:
### Affected hardware version

Bangle 2

### Your firmware version

2v27

### The bug

often, when I leave the clock, a game, settings screen, etc, it displays Loading boot0.
I would not expect that on switching screens.
Sometimes that causes my dark theme to turn light. Sometimes the default clock face changes. Silly chaos just ensues.

I connected to the Web IDE to see if anything would read out when getting this flaw. When i switched out of my clock to the apps drawer and it showed the boot0 message, nothing changed in the IDE display.

bobrippling, 2025-09-14:
To debug further, can you show the output of rinning this in the IDE?

```javascript
require("Storage").list(/boot/)
```

JamoDevNich, 2025-09-24:
Do you use Icon Launcher by any chance? Could you see if the same issue still occurs if you switch to a different launcher