espruino/BangleApps #3999: [app/widget name] Short description of bug

Koell, 2025-09-14:
### Affected hardware version

Bangle 2

### Your firmware version

2v27

### The bug

### Bug
Unable to update, trying to remove app also does not work
- Update App failed, Unexpected response "_"
- Message Icons removal failed, Error: Invalid JSON

[image]

[image]

Koell, 2025-09-15:
I could reproduce the error after resetting by installing an app and before that was finished removing another app

bobrippling, 2025-09-15:
Sounds like it could be a problem with the new app loading code - Espruino/EspruinoAppLoaderCore#67

The code looks ok, but perhaps there's something missing. Are you able to provide the contents of the javascript console from that run?

bobrippling, 2025-09-16:
Cool, so if you hit [F12 (or View > Developer > Javascript Console)](https://developer.chrome.com/docs/devtools/open/), that'll pop it open